ARM-software/CMSIS_5 #1217: arm_biquad_cascade_df2T_compute_coefs_f32 parameter lose const

ua1arn, 2021-06-17:
Last parameter of arm_biquad_cascade_df2T_compute_coefs_f32 really used as pointer to 'const' array data

christophe0606, 2021-06-18:
@ua1arn Thanks for reporting it.

christophe0606, 2021-07-08:
It is corrected in latest commit https://github.com/ARM-software/CMSIS_5/commit/347ca08b52808350c6328cd6a28f42f75dcddaad

ua1arn, 2021-07-08:
Please investigate at this,,,

https://github.com/ARM-software/CMSIS_5/blob/347ca08b52808350c6328cd6a28f42f75dcddaad/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df2T_init_f32.c#L110

Then I pass coeffs array in read0only memory...
Plwase review usage of oarameters and stored fields in arm_biquad_cascade_df2T_instance_f32

christophe0606, 2021-07-09:
@ua1arn You're totally right. This function is wrong since the filter data structure has a constant field for the coefficients.

I am going to rework the API of the function so that it no more try to change the filter coefficients inside the data structure

christophe0606, 2021-07-09:
@ua1arn It should be corrected in https://github.com/ARM-software/CMSIS_5/commit/72dd9172c744bdb4b07c57c1e840b0cd33d723e7

The order of init is now changed.

First we compute the vectorized coefficients:

`arm_biquad_cascade_df2T_compute_coefs_f32(numStages,coefsp,vecCoefsPtr);
`

This API has been changed. It no more use the filter instance (which is not yet initialized) and it is taking a new output argument for the vectorized coefficients.

Then we use this array of new coefficient to initialize the filter.

`arm_biquad_cascade_df2T_init_f32(&this->Sdf2T,numStages,vecCoefsPtr,statep);
`

The coef array in the filter state is read only as it should and no function is attempting to modify it.

closed 2021-07-09 by ua1arn