ARM-software/CMSIS_5 #1221: configLib.cmake is missing

enyg, 2021-06-24:
The CMakeLists.txt files such as CMSIS/DSP/Source/*/CMakeLists.txt have references to configLib. There is no configLib.cmake file in the repository, nor is there any file that defines such a function. As a result, I have been unable to build using CMake. 

There are also references to configApp() which seems to be intended to come from config.cmake. CMake is also unable to find this function.

christophe0606, 2021-06-25:
@enyg Before answering I just want to remind that the official way to build CMSIS-DSP is through the CMSIS-Pack.

That being said, DSP/README.md is explaining how to use the cmake.

There is the need to define a ROOT variable on the cmake command line and it will be used to find the configLib and configApp which are defined in some of the cmake files in the DSP folder.

enyg, 2021-06-28:
Thanks @christophe0606, it seems my mistake was downloading the .pack file rather than cloning the repository. (I downloaded it from the releases [here](https://github.com/ARM-software/CMSIS_5/releases/tag/5.7.0).) Since the folder structure inside the pack looks a lot like the repository and contains a lot of the files, I assumed I was working with the same thing.

closed 2021-06-29 by christophe0606