ARM-software/CMSIS_5 #1223: Remove function definitions from headers

JackNWhite, 2021-06-28:
Hi,

I'm having problems compiling unit tests for Intel in my current project because the CMSIS headers included by the dev board header ultimately define a load functions containing ARM assembly instructions. Additionally, I will want to mock the NVIC functions later on in my tests.

Why are these functions defined in the CMSIS headers and not in a separate source file? Surely it is obvious that this will cause problems? Am I missing something here?

I do not want to maintain separate versions of the headers just because CMSIS is strangely organised.

Can these functions be removed into a separate file?

JonatanAntoni, 2021-06-28:
Hi @JackNWhite,

Thanks for raising this. CMSIS has not been developed with this kind of crossplatform usage in mind.

May I ask you to come up with a proposal how you'd structure the CMSIS header file such that it fits into your use case?

Thanks,
Jonatan

JackNWhite, 2021-06-29:
Hi @JonatanAntoni ,

Thanks for the response and, looking at my initial question, some apologies for my obvious frustration too! Perhaps a little more explanation is needed, too:

### Scenario
I am writing software for a Cortex-M4-based computer, though some components are also intended to work on other devices, including a Cortex-A53 and perhaps in future non-ARM-based computers. My development environment is Linux, or Windows, on an Intel laptop.

To aid in development, unit and integration tests, static analysis and production of the Doxygen documentation should be able to take place on our laptops and via our CI server, both Intel based.

A unit test build, then, comprises:
- Unit source
- All headers referred to by the unit source, both directly and indirectly
- Test source
- Mock sources and headers

### Problems
The above scenario, in particular the test build, only works if the headers contain only passive components: type, function and variable declarations, and preprocessor definitions and commands.

The scenario becomes more difficult if the headers contain function definitions:
1. If the header-defined functions make use of hardware-specifics not available to the test compiler/system, then they cannot be compiled.
2. If the tester wants to mock the header-defined functions, this may be practically impossible, or more at least more difficult.
3. If the header-defined functions themselves call functions defined wholly outside the test build scenario, then these next-level functions must be mocked even if they are not used.

These are compounded by CMSIS reliance on device-specific type(s), e.g. `IRQn_Type`, which is defined in `device.h`.

#### Problem 1 Example
The CMSIS header for GCC (`cmsis_gcc.h`) defines a number of functions which are essentially aliases for assembly language instructions. When this header is included in a source file, the source file cannot be compiled for a non-ARM platform, even when it does not utilise these functions.

The GCC header is included automatically, via, for example, `core_cm4.h`, the Cortex-M4 core header.

#### Problem 2 example
The Cortex-M4 core CMSIS header, is included in the board vendor's CMSIS-compliant `device.h`. `core_cm4.h` implements several interrupt controller functions (`__NVIC_Some_Function()`) and a number of other functions. Because these functions are not simply prototyped, but defined, in the header, it is not readily possible to mock them from a unit which indirectly includes `core_cm4.h`.

### Suggested solution
It is suggested that header in CMSIS be limited to "passive" things, and that "active" components, such as variable definitions and function definitions, be extracted to C source files, which must be compiled along with user sources. This allows a source to include CMSIS headers, but for the "active" components to remain undefined when unused.

#### Example
`core_cm4.h` currently includes the following function definition:
```c
__STATIC_INLINE void __NVIC_EnableIRQ(IRQn_Type IRQn)
{
  if ((int32_t)(IRQn) >= 0)
  {
    __COMPILER_BARRIER();
    NVIC->ISER[(((uint32_t)IRQn) >> 5UL)] = (uint32_t)(1UL << (((uint32_t)IRQn) & 0x1FUL));
    __COMPILER_BARRIER();
  }
}
```
In the proposed solution, `core_cm4.h` will now read:
```c
__STATIC_INLINE void __NVIC_EnableIRQ(IRQn_Type IRQn);
```
... and will be accompanied by `core_cm4.c`, in which will be defined the function body:
```c
__STATIC_INLINE void __NVIC_EnableIRQ(IRQn_Type IRQn)
{
  if ((int32_t)(IRQn) >= 0)
  {
    __COMPILER_BARRIER();
    NVIC->ISER[(((uint32_t)IRQn) >> 5UL)] = (uint32_t)(1UL << (((uint32_t)IRQn) & 0x1FUL));
    __COMPILER_BARRIER();
  }
}
```

A build which includes the new `core_cm4.h` will be able to define its own `__NVIC_EnableIRQ()`, or leave it undefined, or include `core_cm4.c` in order to use the real implementation.

### Notes
I cannot claim to be totally familiar with the whole of CMSIS. I have been aware of it from previous ARM projects, but this is the first where I've had to inspect it. The CMSIS headers I have in my main code repo were provided by the hardware vendor, which expected use of the ARM C compiler. I had to add the CMSIS GCC header myself, which is how I came to this point.
[... 72 lines cut ...]
- __CLZ()
- __LDREXB()
- __LDREXH()
- __LDREXW()
- __STREXB()
- __STREXH()
- __STREXW()
- __CLREX()
- __enable_irq()
- __disable_irq()
- __enable_fault_irq()
- __disable_fault_irq()
- __get_FPSCR()
- __set_FPSCR()
- __get_CPSR()
- __set_CPSR()
- __get_mode()
- __set_mode()
- __get_SP()
- __set_SP()
- __get_SP_usr()
- __set_SP_usr()
- __get_FPEXC()
- __set_FPEXC()
- __FPU_Enable()

JonatanAntoni, 2021-06-29:
Hi @JackNWhite,

I aggree your use case makes perfectly sense.

Unfortunately, we cannot comply with your request. Moving function definitions into source files is not an option for CMSIS. The compiler needs to inline the code when it is used for performance reasons. Lots of the intrinsic functions even rely on being force-inlined and *not* called through a `BL` or `BLX` instruction.

In order to properly isolate your hardware agnostic code you should avoid relying on CMSIS stuff at all. If you can achieve that there should be no issue in unit-testing the code on your development platform. All the defines provided by CMSIS-Core and the vendor device header files are hardware specific anyway. Code that makes use of such hardware specific functions might be very hard to adopt for a different hardware platform or architecture. 

Sorry, if this is not the answer you were looking for. We might check which parts of CMSIS are relevant for hardware-agnostic components of your application. I can imagine to refactor such parts so that you can reuse them without having to duplicate stuff. Other users created their own hardware abstraction layer (HAL) so that software layers above are agnostic of CMSIS.

Cheers,
Jonatan

JackNWhite, 2021-06-30:
Ok, thanks for your time.

closed 2021-06-30 by JackNWhite