ARM-software/CMSIS_5 #1224: Corex-A PL-390 cache functions

ua1arn, 2021-06-28:
As listed in DDI0246H_l2c310_r3p3_trm.pdf, some L2 cache manipulation functions (with physical address as argument) are atimic and not require cache_sync (L2C_Sync() wrapper form CMSIS) for waiting of compete,
In my case, replace called in loop L2C_CleanInvPa for naked write to CLEAN_INV_LINE_PA rise speed of my test application about 10%. I think this because L2C_Sync() wait for some other operaions with cache (quote from document):
`Drain the STB. Operation complete when all buffers, LRB, LFB, STB, and EB, are empty.`

`#if (__L2C_PRESENT == 1)
		int32_t op_size2 = dsize + (addr & (DCACHEROWSIZE - 1U));
		uintptr_t op_addr2 = addr /* & ~(DCACHEROWSIZE - 1U) */;
		do
		{
			// Clean cache by physical address
			L2C_310->CLEAN_INV_LINE_PA = op_addr2;	// Atomic operation. These operations stall the slave ports until they are complete.
			op_addr2 += DCACHEROWSIZE;
			op_size2 -= DCACHEROWSIZE;
		} while (op_size2 > 0);

		__DSB();
		__ISB();
#endif /* (__L2C_PRESENT == 1) */
`

JonatanAntoni, 2021-06-28:
Hi @ua1arn,

Thanks for raising this. Cache maintenance has always been a critical topic. It's hard to have one implementation that fits all needs. The implementation given shall work in most situations. Being aware that it might not be optimal. Optimizing it for certain use cases might break others.

Do you have specific ideas how this implementation shall be optimized?

Cheers,
Jonatan

ua1arn, 2021-06-28:
I propose add cache memory range manipulation functions, looks like SCB_*DCache_by_Addr group in Cortex-M library.

ua1arn, 2021-06-28:
`// Сейчас эта память будет записываться по DMA куда-то
void arm_hardware_flush(uintptr_t addr, int_fast32_t dsize)
{
	//ASSERT((addr % DCACHEROWSIZE) == 0);
	//ASSERT((dsize % DCACHEROWSIZE) == 0);

	if (dsize > 0)
	{
		int32_t op_size = dsize + (addr & (DCACHEROWSIZE - 1U));
		uintptr_t op_addr = addr /* & ~(DCACHEROWSIZE - 1U) */;

		__DSB();

		do
		{
			L1C_CleanDCacheMVA((void *) op_addr);		// записать буфер, кэш продолжает хранить
			op_addr += DCACHEROWSIZE;
			op_size -= DCACHEROWSIZE;
		} while (op_size > 0);

		__DSB();
		__ISB();

#if (__L2C_PRESENT == 1)
		int32_t op_size2 = dsize + (addr & (DCACHEROWSIZE - 1U));
		uintptr_t op_addr2 = addr /* & ~(DCACHEROWSIZE - 1U) */;
		do
		{
			// предполагается, что размер строки L1 и L2 cache равны
			// Clean cache by physical address
			L2C_310->CLEAN_LINE_PA = op_addr2;	// Atomic operation. These operations stall the slave ports until they are complete.
			op_addr2 += DCACHEROWSIZE;
			op_size2 -= DCACHEROWSIZE;
		} while (op_size2 > 0);

		__DSB();
		__ISB();
#endif /* (__L2C_PRESENT == 1) */
	}
}
`

ua1arn, 2021-06-28:
I think this example should be spitted in to two functions named like
`L1_CleanDCache_by_Addr`
and
`L2_CleanDCache_by_Addr`