ARM-software/CMSIS_5 #1226: DSP: Add float 64 bits to basic functions

jeanbie0, 2021-06-30:
Hello,

Some basics functions like Vector Substraction (https://arm-software.github.io/CMSIS_5/DSP/html/group__BasicSub.html) (arm_sub_fxx) are in float 16 and 32 but not in float 64 bits.

I'm doing some of these functions because I had to use it with a compatible double floating point unit processor. Do you know I can push these new files? It'll be more easy to organize my jobs and others people will not have to redo the job.

Thank you

JonatanAntoni, 2021-06-30:
Hi @jeanbie0,

Thanks for getting in touch. Basically, we are open for receiving contribution via Pull-Requests.
I added in @christophe0606 who maintains the DSP part of CMSIS.

Cheers,
Jonatan

christophe0606, 2021-07-01:
Hi @jeanbie0,

Indeed we are missing some f64 code.

The problem is that for f16, f32 and f64 code, we have lot of code duplication for the scalar code. It is basically the same code but with different types and constants.

I am thinking about ways to avoid this. 

But if you don't have a lot of new f64 functions, then I can perhaps merge them even if it means more code duplication.

christophe0606, 2021-09-30:
Hi @jeanbie0,

We have recently added more f64 functions. We are not covering 100% of the existing f32 functions so obviously more will be added in the future.

But we are covering lots of important functions already.

So, I am closing this ticket.

closed 2021-09-30 by christophe0606