ARM-software/CMSIS_5 #1232: CMSIS-DSP, how to build on a non-ARM target?

KammutierSpule, 2021-07-01:
I'm looking to build CMSIS-DSP on a non-ARM target, ie: on a x64-PC for emulation/debug/test functions proposes.

Is it possible? Where should I look at to be able to use CMSIS-DSP library to compile on GCC Linux?

KammutierSpule, 2021-07-01:
Related to https://github.com/ARM-software/CMSIS_5/issues/490 https://github.com/ARM-software/CMSIS_5/issues/1015 https://github.com/ARM-software/CMSIS_5/issues/1149

christophe0606, 2021-07-02:
Hi @KammutierSpule,

First you should define -D__GNUC_PYTHON__ on your compiler command line. It was initially done to build a version of CMSIS-DSP for use for Python but can also be used to compile a version on a host.

Then, if you're building with cmake, your top level cmake should include following 3 lines:

```
set(DSP ${ROOT}/CMSIS/DSP)
list(APPEND CMAKE_MODULE_PATH ${DSP})
add_subdirectory(${DSP}/Source bin_dsp)
```

where root is the path to CMSIS.

KammutierSpule, 2021-07-02:
Hi @christophe0606 

You mention in https://github.com/ARM-software/CMSIS_5/issues/1149#issuecomment-800821845 

> Note that there is a new option. You no more need to set the CMAKE_C_FLAGS to force __GNUC_PYTHON__
> You can just use -DHOST=ON as a cmake option for the CMSIS-DSP.

Does it still applies?

christophe0606, 2021-07-02:
@KammutierSpule  You're right ! I had totally forgotten it. I have added this option recently and it will add the GNUC_PYTHON define.

KammutierSpule, 2021-07-05:
Here is still a question, does that option, `-DHOST=ON`, work also while building to ARM targets?

I would like to compare the C vs ASM implementation of the functions on the ARM target too...

christophe0606, 2021-07-06:
@KammutierSpule It should work but I have not tested.

But there is another possibility. If you use -DAUTOVECTORIZE=ON it will force the use of the pure scalar C versions (useful when we want to let the compiler do the vectorization on the recent M55 core).

Those pure C versions are not using any of the DSP instructions like __SMLAD, __QADD etc ...

The difference with -DHOST=ON is that those pure scalar versions are using the ARM intrinsics __CLZ, __SSAT, __USAT, _ROR.

The host version is replacing those intrinsics with a pure C implementation.

We don't have any ASM in the CMSIS-DSP. But we do use some C intrinsics to target a few specific ASM instructions like the ones above (and the SMLAD for Q15 versions ...)

closed 2021-08-30 by christophe0606

wimalopaan, 2024-02-22:
I encounter problems using this simple host program:

```
#include "CMSIS/DSP/Source/TransformFunctions/TransformFunctions.c"
int main() {
}

```

I get the following errors:

```
In file included from CMSIS/Include/cmsis_compiler.h:54,
from CMSIS/DSP/Include/arm_math.h:310,
from /home/lmeier/Projekte/wmucpp/host/bdc/CMSIS/DSP/Source/TransformFunctions/arm_bitreversal.c:29:
CMSIS/Include/cmsis_gcc.h: At global scope:
CMSIS/Include/cmsis_gcc.h:146:31: error: 'const __cmsis_start()::__copy_table_t __copy_table_start__', declared using local type 'const __cmsis_start()::__copy_table_t', is used but never defined [-fpermissive]
146 |   extern const __copy_table_t __copy_table_start__;
|                               ^~~~~~~~~~~~~~~~~~~~
CMSIS/Include/cmsis_gcc.h:147:31: error: 'const __cmsis_start()::__copy_table_t __copy_table_end__', declared using local type 'const __cmsis_start()::__copy_table_t', is used but never defined [-fpermissive]
147 |   extern const __copy_table_t __copy_table_end__;
|                               ^~~~~~~~~~~~~~~~~~
CMSIS/Include/cmsis_gcc.h:148:31: error: 'const __cmsis_start()::__zero_table_t __zero_table_start__', declared using local type 'const __cmsis_start()::__zero_table_t', is used but never defined [-fpermissive]
148 |   extern const __zero_table_t __zero_table_start__;
|                               ^~~~~~~~~~~~~~~~~~~~
CMSIS/Include/cmsis_gcc.h:149:31: error: 'const __cmsis_start()::__zero_table_t __zero_table_end__', declared using local type 'const __cmsis_start()::__zero_table_t', is used but never defined [-fpermissive]
149 |   extern const __zero_table_t __zero_table_end__;
|                               ^~~~~~~~~~~~~~~~~~
```

using the following compiler call:

`g++ -ICMSIS/DSP/Include  -ICMSIS/Include  -DHOST=ON -O2 -std=c++23  -fconcepts -ftemplate-depth=2048 -fstrict-aliasing -Wstrict-aliasing=1  -Wall -Wextra  /home/lmeier/Projekte/wmucpp/host/bdc/testDSP01.cc --output testDSP01`

Adding `-fpermissive` turns the above to warnings but gives the following errors:

```
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_32_fast_init_f32':
testDSP01.cc:(.text+0x34): undefined reference to `twiddleCoef_rfft_32'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_64_fast_init_f32':
testDSP01.cc:(.text+0x84): undefined reference to `twiddleCoef_rfft_64'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_256_fast_init_f32':
testDSP01.cc:(.text+0xd4): undefined reference to `twiddleCoef_rfft_256'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_512_fast_init_f32':
testDSP01.cc:(.text+0x124): undefined reference to `twiddleCoef_rfft_512'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_1024_fast_init_f32':
testDSP01.cc:(.text+0x174): undefined reference to `twiddleCoef_rfft_1024'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_2048_fast_init_f32':
testDSP01.cc:(.text+0x1c4): undefined reference to `twiddleCoef_rfft_2048'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_rfft_4096_fast_init_f32':
testDSP01.cc:(.text+0x214): undefined reference to `twiddleCoef_rfft_4096'
/usr/bin/ld: /tmp/ccEXwCvV.o: in function `arm_cfft_radix2_init_f32':
testDSP01.cc:(.text+0x997): undefined reference to `twiddleCoef_4096'
/usr/bin/ld: testDSP01.cc:(.text+0x9e3): undefined reference to `armBitRevTable'
/usr/bin/ld: testDSP01.cc:(.text+0xa1e): undefined reference to `armBitRevTable'
/usr/bin/ld: testDSP01.cc:(.text+0xa3e): undefined reference to `armBitRevTable'
/usr/bin/ld: testDSP01.cc:(.text+0xa5e): undefined reference to `armBitRevTable'
/usr/bin/ld: testDSP01.cc:(.text+0xa89): undefined reference to `armBitRevTable'
...
```

How to get around this.

JonatanAntoni, 2024-02-23:
@wimalopaan, please consider raising this in a separate new issue rather than attaching it to an old and closed one. Please take note that CMSIS 5 has been superseded by CMSIS 6 recently. 

You might rather want to raise your questions at https://github.com/ARM-software/CMSIS_6 if related to CMSIS-Core and startup (like in this case). Or at https://github.com/ARM-software/CMSIS-DSP if related to CMSIS-DSP software component. Thanks.