ARM-software/CMSIS_5 #1233: CMSIS-DSP regressions: compiler warning "last line of file ends without a newline"

escherstair, 2021-07-07:
After upgrading from CMSIS-DSP from 1.8.0 to 1.9.0 I noticed an annoying regression.
Some of the files inside CMSIS-DSP doesn't have an empty line at the end anymore.
And so the compiler gives a warning
`warning:  #1-D: last line of file ends without a newline`
Here is the list of these files:

- [CMSIS\5.8.0\CMSIS\DSP\Source\BasicMathFunctions\BasicMathFunctions.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/BasicMathFunctions/BasicMathFunctions.c)
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\arm_cmplx_conj_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_conj_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\arm_cmplx_dot_prod_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_dot_prod_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\arm_cmplx_mag_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mag_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\arm_cmplx_mag_squared_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mag_squared_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\arm_cmplx_mult_cmplx_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mult_cmplx_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\arm_cmplx_mult_real_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mult_real_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\ComplexMathFunctions.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/ComplexMathFunctions.c)
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\ControllerFunctions.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/ControllerFunctions.c)
- [CMSIS\5.8.0\CMSIS\DSP\Source\FilteringFunctions\arm_biquad_cascade_df1_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df1_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\FilteringFunctions\arm_biquad_cascade_df1_init_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df1_init_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\FilteringFunctions\arm_biquad_cascade_df2T_init_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_df2_init_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\FilteringFunctions\arm_biquad_cascade_stereo_df2T_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_biquad_cascade_stereo_df2T_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\ComplexMathFunctions\FilteringFunctions.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/ComplexMathFunctions/FilteringFunctions.c)
- [CMSIS\5.8.0\CMSIS\DSP\Source\StatisticsFunctions\StatisticsFunctionsF16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/StatisticsFunctions/StatisticsFunctionsF16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\StatisticsFunctions\StatisticsFunctions.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/StatisticsFunctions/StatisticsFunctions.c)
- [CMSIS\5.8.0\CMSIS\DSP\Include\arm_const_structs_f16.h](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/arm_const_structs_f16.h)
- [CMSIS\5.8.0\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix2_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix2_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix4_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\TransformFunctions\arm_rfft_fast_init_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\TransformFunctions\arm_cfft_radix8_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix8_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\TransformFunctions\arm_bitreversal_f16.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/arm_bitreversal_f16.c) - this is a new file in CMSIS-DSP 1.9.0
- [CMSIS\5.8.0\CMSIS\DSP\Source\TransformFunctions\TransformFunctions.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/TransformFunctions/TransformFunctions.c)

Can you fix all of these warnings?

christophe0606, 2021-07-07:
@escherstair I'll correct it. Which compiler are you using ? I don't have those warnings with clang.
I want to track it for the future.

escherstair, 2021-07-07:
ARM Compiler 5 from Keil uVision 5.34

christophe0606, 2021-07-08:
It should be corrected in commit https://github.com/ARM-software/CMSIS_5/commit/347ca08b52808350c6328cd6a28f42f75dcddaad

closed 2021-08-30 by christophe0606