ARM-software/CMSIS_5 #1234: CMSIS-DSP: compiler warning "single-precision operand implicitly converted to double-precision"

escherstair, 2021-07-07:
I've just upgraded CMSIS-DSP from 1.8.0 to 1.9.0 and this brings to new compiler warnings.
File [arm_rotation2quaternion_f32.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c) gives several "single-precision operand implicitly converted to double-precision".
The reason is that single-precision constant numbers should be written with 'f' suffix (i.e. `1.0f`).
Here are the lines with the warning:

- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L183
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L184
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L185
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L192
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L193
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L195
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L201
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L202
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L205
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L210
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L211
- https://github.com/ARM-software/CMSIS_5/blob/08b739316857fb4c6f801f0ed4d5deded62cb297/CMSIS/DSP/Source/QuaternionMathFunctions/arm_rotation2quaternion_f32.c#L215

Can you fix these warnings, please?

christophe0606, 2021-07-07:
@escherstair  Thanks for reporting it. It should have been caught.
It looks like -Wall -Wextra  -Werror is not enough to detect it.

I'll correct it.

christophe0606, 2021-07-08:
@escherstair I have started to work on those issues of float promotion. It will take more time than I had anticipated because I want to correct all issues and the float16 code has a lot more because of ambiguity with types __fp16 and _Float16.

You have not seen those issues because float16 is disabled when building CMSIS-DSP with ARM Compiler 5.

Thanks for reporting this problem which is not detected with -Wall -Wextra on gcc and clang. I have added the missing option to track this issue in the future.

escherstair, 2021-07-08:
@christophe0606 changing the compiler from ARM Compiler 5 to 6 is one of the next tasks on my roadmap.
So, thank you for your job.
I hope that the release with these fixes will be released soon.

christophe0606, 2021-07-08:
It should be corrected (among other related issues) in commit https://github.com/ARM-software/CMSIS_5/commit/347ca08b52808350c6328cd6a28f42f75dcddaad

closed 2021-08-30 by christophe0606