ARM-software/CMSIS_5 #1235: Performance question: DSP, block vs intrinsics processing

KammutierSpule, 2021-07-07:
I noticed that the DSP processing routines are mostly blocksize processing.

I'm wondering what would be the differences / advantages of doing it as block processing vs an algorithm that is using a sequential use of intrinsics (eventually over a block) processing ?

Using this DSP block functions, each one will add a for loop vs I can implement with just one for loop my algorithm.

Any guidance over this question?

christophe0606, 2021-07-08:
Hi @KammutierSpule ,

I am not sure I understand the question :-)

Block processing is used for filters to amortize the cost of managing a circular buffer (state). If it was done for small blocks (or each sample) it would be too costly since there is no specific circular addressing mode in the instruction set.

Now, if you are doing a lot more in a loop than just filtering then perhaps the management of the circular buffer would become less critical and you could perhaps work with smaller blocks.

So, perhaps, the fusion of several loops into just one may help to have a more standard implementation (sample based and not block based).

But, it really depends on your final application and benchmarks. 

On very performance critical apps, you really need to do a block processing.

KammutierSpule, 2021-07-08:
Thanks Christophe, I will better elaborate. I'm referring to create my own algorithm, using BasicMathFunctions (and not using the other functions/filters on the library).

```
// dst = A * B + B
void MyAlgo1(   const q31_t * pSrcA, const q31_t * pSrcB, q31_t * pDst, uint32_t blockSize )
{
arm_mult_q31( pSrcA, pSrcB, pDst, blockSize );
arm_add_q31( pDst, pSrcB, pDst, blockSize );
}
```

vs

```
// dst = A * B + B
void MyAlgo2(   const q31_t * pSrcA, const q31_t * pSrcB, q31_t * pDst, uint32_t blockSize )
{
  while (blockSize > 0)
  {
    q31_t B = *pSrcB++;

    /* Multiply inputs and store result in destination buffer. */
    out = ((q63_t) *pSrcA++ * B ) >> 32;
    out = __SSAT(out, 31);
    out = out << 1U;

    /* Add and store result in destination buffer. */
    *pDst++ = __QADD( out, B );

    blockSize--;
  }
}
```
I'm wondering which one will perform better, since the BasicMathFunctions has unroll loops and I'm not sure how much it will perform better when doing it as MyAlgo1 or if I should get better performance (perhaps some better register optimization by compiler) by doing MyAlgo2 ?

christophe0606, 2021-07-08:
@KammutierSpule In the first case, the compiler will likely not be able to fusion the two loops.
So, it will lead to additional memory accesses. 

In the second case, the loops are no more unrolled. But the only advantage of the unrolling for functions as simple as the arm_add_q31 and arm_mul_q31 is to have more work to do in the loop so that the loop management overhead becomes just a small part of the processing.

Since your second case is doing more work inside the loop, the unrolling is probably not needed anymore. Thus, I assume your second version should be more efficient because it is doing less memory accesses.

But you have to benchmark because it is highly compiler dependent. Your compiler may unroll the loop and do lot of other things.

In a full application, I don't expect using 1 or 2 should make a big difference. You'll probably be spending most of your cycles in more heavier processing somewhere else.

KammutierSpule, 2021-07-08:
ok I think on my case, it will be better to do 2 (as there will be more complicated operations)
Thanks Christophe for the great support!

closed 2021-07-08 by christophe0606