ARM-software/CMSIS_5 #1236: arm_relu6_s8 as hard coded comparison value of 6

uu-praveeng, 2021-07-09:
Hi,

In the function arm_relu6_s8, I have seen that comparison is done with 6 but actually it should be done with scale value of 6. Let me know if I am missing anything.

Is there a generic relu implementation for S8 other than relu6? If not, Are there any plans to add that?

Thanks,
Praveen.

uu-praveeng, 2021-07-09:
I for s8 relu I guess I could use arm_relu_q7. But, please check on the arm_relu6_s8

felix-johnny, 2021-07-12:
@uu-praveeng  Thanks for the question.
 arm_relu_q7() is the generic relu function but I am not sure if would solve the case that you are after i.e, scaled limits. But, that said, in all of the functions(conv, FC, dw conv etc..), activation is fused and you can set the min and max. That would result in better performance as well.

This is the closest I could find on relu6 https://www.tensorflow.org /api_docs/python/tf/nn/relu6 . Could you send a reference where you see that the 6 is to be scaled?

uu-praveeng, 2021-07-12:
Thanks, Felix. Activations as part of FC and conv will be the way to go for relu with limits.  

I have no use case as of today. But from the name arm_relu6_s8. I feel 6 should be scaled to the s8 value and then used. I will revert back when I have an actual use case.

We could ignore this as of now.

closed 2021-07-12 by uu-praveeng