ARM-software/CMSIS_5 #1237: CMSIS-DSP: improve documentation on how to include needed tables only

escherstair, 2021-07-09:
[Here](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP#compilation-symbols-for-tables) I can see that some new compilations symbols have been introduced to avoid including all the tables if they are not needed.

> If no new symbol is defined, everything will behave as usual. If ARM_DSP_CONFIG_TABLES is defined then the new symbols will be taken into account.
> 
> Then you can select all FFT tables or all interpolation tables by defining following compilation symbols:
> - ARM_ALL_FFT_TABLES : All FFT tables are included
> - ARM_ALL_FAST_TABLES : All interpolation tables are included

From the above lines, it seems that once ARM_DSP_CONFIG_TABLES has been defined, it's enough to select which tables are needed.
I think this is not true, unfortunately, because I tried and it didn't work.

So I looked inside [cmsisdspconfig.py](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/cmsisdspconfig.py) and I see
https://github.com/ARM-software/CMSIS_5/blob/72dd9172c744bdb4b07c57c1e840b0cd33d723e7/CMSIS/DSP/cmsisdspconfig.py#L292-L298

If I'm right this means that I need to define **three** symbols:
- ARM_DSP_CONFIG_TABLES                    to enable the "selective" build
- ARM_FFT_ALLOW_TABLES                      so that the needed fft tables **could be** included (if at least one of them is needed)
- ARM_FAST_ALLOW_TABLES                    so that the needed interpolation tables **could be** included (if at least one of them is needed)
then the other symbols of the needed tables.

The necessity of ARM_FFT_ALLOW_TABLES and/or ARM_FAST_ALLOW_TABLES seems confirmed looking into the source code.
As an example
https://github.com/ARM-software/CMSIS_5/blob/72dd9172c744bdb4b07c57c1e840b0cd33d723e7/CMSIS/DSP/Source/FastMathFunctions/FastMathFunctions.c#L29

If this is the case, I suggest to improve [README.md](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP#compilation-symbols-for-tables) adding this clarification.

christophe0606, 2021-07-12:
@escherstair Indeed, the ALLOW options are used because a build may not include the FFT code and in that case the FFT tables must not be included.

I'll update the README to advise people to use the Python configuration script to generate the -D options for the command line. It is the simplest otherwise knowing which option to use require to know how the algorithm is implemented (which tables are needed).

escherstair, 2021-07-12:
@christophe0606 if you'll give advise to use the Python script, please specify that the -D options for the command line are generated if the build method is set to Make

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606