ARM-software/CMSIS_5 #1240: CMSIS-DSP: loop unroll

escherstair, 2021-07-12:
[Python script](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/cmsisdspconfig.py) to configure CMSIS-DSP library has on option named "LOOPUNROLL".
It's set to True by default, and this adds `-DARM_MATH_LOOPUNROLL` to the command line build.
As far as I can see [here](https://www.keil.com/pack/doc/CMSIS/DSP/html/index.html)

- ARM_MATH_LOOPUNROLL:
  Define macro ARM_MATH_LOOPUNROLL to enable manual loop unrolling in DSP functions

Since it has a True default in the Python script, I could think that it should be enabled in common situations.
But I can be wrong.

Where can I find detailed info on "LOOPUNROLLING"?
When should I enable/disable it?
Does it depend on the device?

Thanks in advance

happyTonakai, 2021-08-11:
https://developer.arm.com/documentation/dui0472/c/compiler-coding-practices/loop-unrolling-in-c-code this should help

escherstair, 2021-08-11:
Thank you.
This is quite clear.

christophe0606, 2021-08-31:
@escherstair Sorry for late answer.
It is highly compiler dependent. In theory, this CMSIS-DSP option should no more be needed because compilers are getting very good at unrolling loops.

In practice, and mainly with fixed point functions, we observe that this option is still useful and the manually unrolled versions have better performances.

But you have to benchmark to be sure.

closed 2021-09-30 by christophe0606