ARM-software/CMSIS_5 #1241: ARMCM3 startup and linker

AbhijitInamdar5, 2021-07-12:
Hi, I was using startup_ARMCM3.S (v2.0.0) for my custom machine in qemu-2.10. Now, I updated my qemu-2.10 to qemu-4.2.0 and using the same startup script for my custom machine but it wont work. It is giving me a blank screen. I have tried with all the new scripts (both .ld and .S) on my custom machine it not working as it did in the qemu-2.10. Is there a correct version for the qemu-4.2.0?

JonatanAntoni, 2021-07-12:
Hi @AbhijitInamdar5,

I am sorry but we cannot support qemu officially. You may want to try raising this question with the qemu community to figure out the difference between the major versions. Would be nice if you could post your concludion here one you got a solution. Just in case other CMSIS users are interested.

Thanks,
Jonatan

AbhijitInamdar5, 2021-07-12:
Apart from that which version of startup and script do you suggest me for compiling?

JonatanAntoni, 2021-07-12:
I can only recommend to use the latest version from either master branch (stable, release) or from develop (unstable). You could give the C-Startup file a try. But if the memory layout of your emulated device doesn't fit to the linker script that won't change much. The linker scripts provided are aligned with the memory layout of Arm Fixed Virtual Platform (FVP) models.

AbhijitInamdar5, 2021-07-20:
Hi,

How do I use print in startup script inside reset_handler loops?

I tried something like this
Reset_Handler: 
....
......
ldr     r6, =string

.data
string: .asciz "step0\n"

but this is giving me compilation error, Error: .size expression for Reset_Handler does not evaluate to a constant

Is it correct way?

JonatanAntoni, 2021-07-20:
Hi @AbhijitInamdar5,

I don't really get your question. Am I right that you are trying to implement specific behaviour in GNU Assembly? In this case I'd kindly like to point you to the GCC Assembly community.

Cheers,
Jonatan

AbhijitInamdar5, 2021-07-20:
In the startup_ARMCM3.S inside Reset_handler:   I want to have some print lines example: in c-program printf() function how do I add such kind in .S file ?

JonatanAntoni, 2021-07-21:
Basically, you need to load the function call registers and jump to `printf()`. This may or may not work in the `Reset_Handler` depending on the C-library you are using and the channel `printf` writes to. Please bare in mind that `Reset_Handler` is very early stage in system startup. The C-Library runtime environment has not been executed in this stage! Hence it it might be undefined behaviour to call into the C-Library at this point in time. You need to consult the lib's documentation.

I am sorry that I cannot provide you details here. Your questions are rather generic and not directly related to CMSIS at this point.

AbhijitInamdar5, 2021-07-21:
I'm a newbie. Sorry for my bad communication. I will try to rephrase my question in Reset_Handler my idea was to print some string in between the loop. Something like this :

	ldr	r4, =__copy_table_start__
	ldr	r5, =__copy_table_end__

.L_loop0:
	cmp	r4, r5
	bge	.L_loop0_done
	ldr	r1, [r4]
	ldr	r2, [r4, #4]
	ldr	r3, [r4, #8]
	ldr     r1, =string

.L_loop0_0:
	subs	r3, #4
	ittt	ge
	ldrge	r0, [r1, r3]
	strge	r0, [r2, r3]
	bge	.L_loop0_0

	adds	r4, #12
	b	.L_loop0
	ldr     r1, =string

.L_loop0_done:

.data
string: .asciz "step0\n"

I'm getting error when I compile so is this the right way?

AbhijitInamdar5, 2021-07-25:
Hi, How are the stack size, heap size dependent on the Flash length and RAM length? How is that calculation made?

JonatanAntoni, 2021-07-26:
Hi @AbhijitInamdar5,

The reserved/used memory for stack and heap need to be configured explicitly, depending on the used toolchain. In case of GCC this can be done in the Linker script, e.g. https://github.com/ARM-software/CMSIS_5/blob/develop/Device/ARM/ARMCM3/Source/GCC/gcc_arm.ld#L53-L54.

Using C library functions like printf needs to be considered when reserving stack space.

Cheers,
Jonatan

AbhijitInamdar5, 2021-07-26:
Depending on the toolchain used can you elaborate? I'm using q2-update-2017.

AbhijitInamdar5, 2021-07-27:
Why was these if statements were removed https://github.com/ARM-software/CMSIS_5/blob/5.3.0/Device/ARM/ARMCM3/Source/GCC/startup_ARMCM3.S#L131-L249

While compiling gcc before I used  to use arm-none-eabi-gcc -g hello_world.c startup_ARMCM3.S -mthumb -mcpu=cortex-m3 -D__STARTUP_CLEAR_BSS -D__START=main -Os -flto -ffunction-sections -fdata-sections --specs=nano.specs --specs=rdimon.specs -Wl,--gc-sections -Wl,-Map=main.map -T gcc_arm.ld -o main.axf.  

How should the new one be?  Is there any particular toolchain version is should have?

JonatanAntoni, 2021-07-28:
Hi @AbhijitInamdar5,

We are testing CMSIS with latest toolchain versions, typically. You can find the test environment documented in the [dockerfile](https://github.com/ARM-software/CMSIS_5/blob/b7e9379aab53d173b3b3035ec7253e91917d57f7/docker/dockerfile#L39).

At some point we reduced the complexity of the startup code. The latest version uses the multiple sections scheme only. In order to have this working one need to use the provided linker script. The linker script exports the required copy- and zero-tables. Please note that the zero-table is empty by default, i.e. clearing the bss section is left to newlib.

Again, I want to encourage you to get in touch with the GCC community. Your questions are only losely related to CMSIS. There are other experts around who are much more familiar with the GCC specifics you are looking for.

Cheers,
Jonatan

AbhijitInamdar5, 2021-07-28:
Similarly, if I try to find in cmsis-4.3.0 where is it documented?

JonatanAntoni, 2021-07-28:
CMSIS 4 is at https://github.com/ARM-software/CMSIS

AbhijitInamdar5, 2021-07-28:
My question was in CMSIS 4 is the test environment the same as CMSIS_5? I did not see anything like docker in that repository ? was it even documented ?

JonatanAntoni, 2021-07-28:
No, CMSIS 4 is fairly old and not maintained anymore. Adding docker files has been done recently to CMSIS 5. Testing of CMSIS 5 unfortunately requires non-public stuff at the moment hence it is somehow hidden.

AbhijitInamdar5, 2021-07-28:
Then what I get from these conversation is the compatibility issue with my toolchain being used! Thanks

AbhijitInamdar5, 2021-08-10:
Hi, in the latest linker and startup there is no _bss_start_ and _bss_end_! then how do we define it ? because I tried to debug the startup its looping around the https://github.com/ARM-software/CMSIS_5/blob/develop/Device/ARM/ARMCM3/Source/GCC/startup_ARMCM3.S#L92-#L96. Its not able to exit

JonatanAntoni, 2021-09-06:
Hi @AbhijitInamdar5,

Actually, the Linker symbols `bss_start` and `bss_end` *are* defined in the latest Linker script, e.g. https://github.com/ARM-software/CMSIS_5/blob/cbfe6e37b2d1282f754edc6af4d22bb1073dc4b2/Device/ARM/ARMCM3/Source/GCC/gcc_arm.ld#L244-L253. 

The loop you are refering to is the copy table not the zero table. bss sections are zero initialized by the zero table. The copy table shall list all data sections to be copy initialized from flash. The loop shall stop after iterating over all data sections. If this doesn't work for you we'd need to deeply dig into the details. Please carefully single step the assembly code and verify the register values, `r4` and `r5` are used for `__copy_table_start__` and `__copy_table_end__` respectively. `r4` shall be incremented by 12 for each data section. Once `r4` becomes equal to `r5` the loop shall stop.

Cheers,
Jonatan