ARM-software/CMSIS_5 #1242: 5.8.0 ARMCC compat header known issue list details addition

0xc0170, 2021-07-12:
I am testing CMSIS 5.8.0 update with Mbed OS. The update cause some errors that I spent some time to find out why and how to fix them. Known issue 5.8.0 lists compat header removal from CMSIS and if a user needs it, they should include it explicitly - including it prior any cmsis file.

As you know, Mbed OS has lot of dependencies and inclusion is not trivial. Our retarget looks like:

```
include std lib headers
include some Mbed OS headers
...code..
include compat header if it's ARMCC
...code...
```

When I updated to 5.8.0, I got this errors first:
 
```
ARMCLANG\bin\..\include\arm_compat.h:67:1: error: functions that differ only in their return type cannot be overloaded
__disable_irq(void) {
^
./cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_armclang.h:750:27: note: previous definition is here
__STATIC_FORCEINLINE void __disable_irq(void)
                     ~~~~ ^
In file included from .\platform\source\LocalFileSystem.cpp:26:
C:\Keil_v5\ARM\ARMCLANG\bin\..\include\arm_compat.h:114:1: error: redefinition of '__enable_irq'
__enable_irq(void) {
^
./cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_armclang.h:737:27: note: previous definition is here
__STATIC_FORCEINLINE void __enable_irq(void)
```

First, I was like if this is causing a problem, CMSIS should provide definitions we need. I remove compat header completely from the code file. This resulted in some definitions missing (`__semihost` missing). Ok we need compat but how to address the error. I realized there is workaround I can try to apply, move the compat header to the top of the code file. compat still provides some functionality not covered by CMSIS.

What I would find useful in the known issue list:
- add a section about redefinitions errors people might get if they do not apply the workaround - It should answer why the order matters and users can find this if they get an error . As this can be breaking change for applications, it was for me.
I would add something like: "if compat header is not included prior cmsis headers, it might cause redefinitions errors (or any other errors that are known to be caused). For instance, `error: redefinition of '__enable_irq'
__enable_irq(void)`.  This is because .... (add some details, I would find it useful to know why order matters). 
- it might be useful to at least mention there is overlap between CMSIS and compat or is it well known - there is a mention of CMSIS defining more secure functions instead - what are these? Can we get what we should use from CMSIS ? (#1211 provides some useful details what combat provides and CMSIS does not). I have `__semihost` missing in some files so I checked if this is in CMSIS. I found out it is not, so I need to add combat file in multiple places (still getting through errors as there is lot of files involved).

JonatanAntoni, 2021-07-12:
Please review the updated "Known Issues" section of CMSIS 5.8.0 release notes.

0xc0170, 2021-07-19:
Thanks, I'll close this as resolved.

closed 2021-07-19 by 0xc0170