ARM-software/CMSIS_5 #1244: CMSIS-DSP: warning "statement is unreachable"

escherstair, 2021-07-13:
When some of the tables are not included, as described [here](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP#compilation-symbols-for-tables), the compiler gives warnings "statement is unreachable".
As an example, when none of the f64 tables is included:
https://github.com/ARM-software/CMSIS_5/blob/999d63b1476c53374ff65b23710d8f3d2c482323/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f64.c#L337-L338
This is because of
https://github.com/ARM-software/CMSIS_5/blob/999d63b1476c53374ff65b23710d8f3d2c482323/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f64.c#L333-L334
But the line
https://github.com/ARM-software/CMSIS_5/blob/999d63b1476c53374ff65b23710d8f3d2c482323/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f64.c#L334
should be changed to
`break;`
so that
https://github.com/ARM-software/CMSIS_5/blob/999d63b1476c53374ff65b23710d8f3d2c482323/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f64.c#L337-L338
is executed and `ARM_MATH_ARGUMENT_ERROR` is returned.

christophe0606, 2021-07-13:
@escherstair Thank for reporting this. 

With cmake build when none of the table is included, the file is not compiled.
But with Pack or Make it is not the case so this should be corrected.

I'll check other inits in case similar issue is present.

christophe0606, 2021-07-19:
Corrected in commit https://github.com/ARM-software/CMSIS_5/commit/6df0592c465a4aca479cfd597786b3b6307b99ff

closed 2021-08-30 by christophe0606