ARM-software/CMSIS_5 #1257: ERRORS with C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S

juliobc, 2021-07-23:
Hi, I update to mdk 5.35 just now. I use CMSIS DSP and ARMClang and I find the next problem:

Build started: Project: test
*** Using Compiler 'V6.16', folder: 'C:\Keil_v5\ARM\ARMCLANG\Bin'
Build target 'test'
assembling irq_armv7m.S...
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(1): error: A1167E: Invalid line start 
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(2): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(3): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(4): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(5): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(6): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(7): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(8): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(9): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(10): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(11): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(12): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(13): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(14): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(15): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(16): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(17): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(18): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(19): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(20): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(21): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(22): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(23): error: A1159E: Label missing from line start
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(24): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(27): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(29): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(31): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(32): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(33): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(34): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(35): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(37): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(38): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(39): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(41): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(43): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(45): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(46): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(47): error: A1167E: Invalid line start 
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(48): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(51): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(52): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(53): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(54): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(57): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(58): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(59): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(60): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(61): error: A1137E: Unexpected characters at end of line
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S(62): error: A1167E: Invalid line start 
".test.axf" - 50 Error(s), 0 Warning(s).
Target not created.

Any solution?
Regards

juliobc, 2021-07-23:
If I copy C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\ARM\irq_armv7m.s to C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.8.0\CMSIS\RTOS2\RTX\Source\GCC\irq_armv7m.S (overwrite), then it compiles.

I think some dependency must be wrong

regards

jkrech, 2021-07-23:
Can you please check the Options for Target settings for the Assembler in uVision.
The "Assembler Option:" drop down should be configured to "armclang (Auto Select)".
The background for that is the change in the RTX5 source code variant using the GNU assembler syntax in "irq_armv7m.S" as you can also see from the above file path: RTX\Source\GCC\ .
The above option will detect the GASM syntax and translate accordingly.
The root cause is the transition from Arm Compiler 5 assembler to Arm Compiler 6 assembler.
Sorry for the trouble.
Regards,
Joachim

juliobc, 2021-07-23:
Yes!! that was the problem!
With "armclang (Auto Select)" option it compiles perfectly (before it was "armasm (ARM Syntax)")

Thank you very much

closed 2021-07-23 by juliobc