ARM-software/CMSIS_5 #1258: DSP: f32 compensated summation in arm_mat_mult_f32

escherstair, 2021-07-23:
I ask this because I've just found a real case of high numerical error in the sum of a finite number of floating `f32` numbers.
It comes from the multiplication of two matrices with `f32` values (that involves products and sums).
When you sum two `f32` values with different magnitudes, the obvious approach `a + b` gives rounding error.
And when you mutiply two matrices this can happen.

I saw this on another architecture (not Cortex-M) and I solved applying [Kahan](https://en.wikipedia.org/wiki/Kahan_summation_algorithm) algorithm.

I wonder if the CMSIS-DSP implementation suffer of this potential issue, or not.
I would have never believe to meet this kind of issue in real life, but it happened to me.

christophe0606, 2021-07-26:
@escherstair float can be very tricky and since CMSIS-DSP is focusing on speed I don't expect its algorithms to cover all possible cases. In some cases it will be required to pre-scale inputs, condition matrices or ... use a more accurate algorithm.

I am interested in the values for your matrices.

I'll tag this issue as enhancement because I don't expect Kahan summation to be added to CMSIS-DSP in short term.

escherstair, 2021-07-26:
@christophe0606 I agree that floating point representation is tricky.
And I understand that most of the time speed is the main target.
Algorithms as Kahan (or similar) should always be optional, so that the user is responsible to use them, if needed.

Here is two vectors in float32 that multiplied and accumulated should give `0.0f,` but on my architecture (not ARM) gives `32.0f`.
`A = [55595.0f, 55559.0f, 8805.384f, 9023.475f, 55595.99f, 55559.99f, 8805.54f, 9023.635f]`
```
B = [-8805.384f,
-9023.475f,
55595.0f,
55559.0f,
-8805.54f,
-9023.635f,
55595.99f,
55559.99]
```
They come from bigger matrices, but they are the relevant part to show what happens.

closed 2021-09-30 by christophe0606