ARM-software/CMSIS_5 #1261: fixed-point biquad filter gives wrong results when postShift is not zero

happyTonakai, 2021-07-26:
Hi. I'm using the fixed-point version of biquad IIR filter. Here's a minimum working example:
```
#include <stdio.h>
#include "arm_math.h"
void main()
{
	uint8_t numStages = 3;
	arm_biquad_casd_df1_inst_q31 S;
	q31_t pCoeffs[15] = { 885525418, 1776285333, 890770255, -691745680, -91052396, 885525418, 1771023633, 885508474, -793554000, -420511040, 885525418, 1765843546, 880328308, -1065053360, -1299101426 };
	q31_t pState[12] = { 0 };
	int8_t postShift = 0;
	arm_biquad_cascade_df1_init_q31(&S, numStages, pCoeffs, pState, postShift);

	int8_t blockSize = 10;
	q31_t input[20] = { 0 };
	printf("input:\n");
	for (int i = 0; i < blockSize * 2; ++i)
	{
		input[i] = (i * 100000);
		printf("%d ", input[i]);
	}
	q31_t output[20] = { 0 };

	arm_biquad_cascade_df1_fast_q31(&S, input, output, blockSize);
	arm_biquad_cascade_df1_fast_q31(&S, input + blockSize, output + blockSize, blockSize);

	printf("\noutput signal:\n");
	for (int i = 0; i < blockSize * 2; ++i)
	{
		printf("%d ", output[i] );
	}
	printf("\n");
}
```
When I set `postShift = 0` it's ok and I got the same results as MATLAB gives. 
```
output signal:
0 7012 47766 144468 266186 365746 454058 558270 663568 758670 857726 961166 1060058 1158516 1259952 1360172 1459194 1559542 1659964 1759542
```
However, I notice that the input signal must lie in the range of [-0.25, 0.25) to avoid overflow, so I shifted the input and set `postShift` by
```
int8_t postShift = 2;
...
input[i] = (i * 100000) >> 2;
```
But this gives me a wrong results:
```
output signal:
0 112176 364552 426032 600576 995152 691024 1231112 1793128 267184 2828136 2338360 -1993736 8682616 -1119376 -6613352 28901104 -27364744 -783680 83529760
```
If I shift the output manually by
```
int8_t postShift = 0;
...
input[i] = (i * 100000) >> 2;
...
printf("%d ", output[i] << 2 );
```
I got a right answer but in low-precision:
```
output signal:
0 7008 47768 144464 266184 365752 454048 558280 663576 758664 857712 961168 1060064 1158520 1259944 1360176 1459192 1559536 1659976 1759544
```
Please tell me whether I'm using `postShift` correctly. Thank you in advance.

BTW, `arm_biquad_cascade_df1_fast_q15` gives  a similar result.

christophe0606, 2021-07-26:
@happyTonakai The documentation is confusing. 

In arm_biquad_cascade_df1_q31.c it is written:

`In order to avoid overflows completely the input signal must be scaled down by 2 bits and lie in the range [-0.25 +0.25).`

But in arm_biquad_cascade_df1_init_q15.c, it is written

`postFix : Shift to be applied to the accumulator result. Varies according to the **coefficients** format
`

And when we look at the code, we see that the state is updated with:

```
Xn2 = Xn1;
Xn1 = Xn;
Yn2 = Yn1;
Yn1 = (q31_t) acc;
```

The role of postFix is to force acc to q31 after the update.
If the input signal was scaled and postFix different from 0, then the state update would mix values with different formats.

So the comment above is correct : it is the coefficients which must be scaled when you change the postFix value.

The other comment about scaling the signal between [-0.25, 0.25) is also correct but it is not always needed and not always enough : the sum which can saturate is involving also the internal state. 

So, the saturation can come either from the input signal and/or the coefficients of the filter (resonance ...)

Conclusion : if you detect a saturation you'll have to either scale the signal and / or scale the coefficients and change the postFix value.

happyTonakai, 2021-07-27:
Thanks a lot for your response. I got the usage of `postShift` now. It handles the case that filter coefficients are greater than 1. In this case we scale down the coefficients but `a_0` remains to be 1. Thus a simple post shift back on the output doesn't work.

In the more simple case that input signal to be shifted to the range of [-0.25, 0.25), another post shift should be applied separately. `postShift` is not for this case.

closed 2021-07-27 by happyTonakai