ARM-software/CMSIS_5 #1262: Convolution error

11223uhh, 2021-07-26:
My code looks something like this

arm_convolve_HWC_q7_basic
													
arm_relu_q7
arm_maxpool_q7_HWC
arm_convolve_HWC_q7_fast(img_buffer2,
										
arm_relu_q7
The output of the first convolution is fine, but the output of the second convolution is off by a very large amount

11223uhh, 2021-07-26:
The first channel of the output of the second convolution is always zero

newton1010, 2021-08-11:
I have the same problem in "arm_convolve_HWC_q15_fast" function with ARM_MATH_DSP macro.

felix-johnny, 2021-08-13:
@11223uhh @newton1010 .. Is this a new change? From what I see, 68e9b6013 is the latest change in that area. Could you try without(revert) it?

felix-johnny, 2021-09-16:
@11223uhh Could you please update if you have tried the suggestion above?