ARM-software/CMSIS_5 #1266: irq_armv7m: missing fpu selection

0xc0170, 2021-07-28:
The file https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/ARM/irq_armv7m.s contains only one FPU selection:

```
                IF       ({FPU}="FPv4-SP")
FPU_USED        EQU      1
                ELSE
FPU_USED        EQU      0
                ENDIF
```

There are multiple options of FPU for v7m as I know of. For instance I found it failing for cortex-m4f.

It should be (if I got the FPU flags correct):

```
                IF       ({FPU}="FPv4-SP") || ({FPU}="VFPv4_SP_D16") || ({FPU}="FPv5-SP") || ({FPU}="FPv5_D16")
FPU_USED        EQU      1
                ELSE
FPU_USED        EQU      0
                ENDIF
```

jkrech, 2021-07-28:
Playing with the command line options I could not create FPv4-SP in the info {FPU} but using --fpu option I was able to see VFPv4_D16

 --cpu  | --fpu | {FPU} 
-------|--------| -------
Cortex-M4.fp.sp |   | VFPv4_SP_D16
Cortex-M4.fp     |   | VFPv4_SP_D16
Cortex-M4         | FPv4-SP | VFPv4_SP_D16
Cortex-M4         | VFPv4_D16  | VFPv4_D16
Cortex-M7.fp.sp |   | FPv5-SP 
Cortex-M7.fp.dp |   | FPv5_D16

JonatanAntoni, 2021-07-29:
Party addressed in #1267

0xc0170, 2021-07-29:
> Playing with the command line options I could not create FPv4-SP in the info {FPU} but using --fpu option I was able to see VFPv4_D16

This would be the correct condition (taking {FPU} from the table above)?

`                IF       ({FPU}="VFPv4_D16") || ({FPU}="VFPv4_SP_D16") || ({FPU}="FPv5-SP") || ({FPU}="FPv5_D16")`

JonatanAntoni, 2021-07-29:
@0xc0170, please check with @GorgonMeducer about getting this into a PR. Thanks!