ARM-software/CMSIS_5 #1268: The arm Convolve HWC Q7 RGB takes too long. Procedure

11223uhh, 2021-08-01:
I use lenet model, which runs in STM32F407 with the main frequency of 168M. The total time of the model is 100ms, among which arm Convolve HWC Q7 RGB takes 75ms. Is there any way to improve it

felix-johnny, 2021-08-02:
@11223uhh Could you give some more details of the model that you used for the measurements ? I am looking for information like,
1. input and filter dimensions of the layers that take up most of the cycles , (Name of the operator as well)
2. Compiler and its version.
3. Options passed on to the compiler

11223uhh, 2021-08-04:
1 The input is 50*50*3  the convolution kernel is 5*5*8
2.I used the KEIL5 simulation for time testing, which is version 5.26.2.0

felix-johnny, 2021-08-13:
@11223uhh Thanks for the info. Unfortunately it is not enough yet to give a useful suggestion.

 A dimension of 50x50x3 has multiple unaligned accesses(input channel not a multiple of 4) , so you can expect it to take more cycles than a layer with say 4 input channels. But, that said, if it is a major contributor depends on how the rest of the layers that follow are. If you have say just one or two layers without as many MACs following it, then yes, it will be a major contributor. It is hard to say this without more information. Hence my first questions about different layers and their shapes. That includes output shape as well. Or you can calculate the number of MACs and give that information instead. 

 Please answer question 2 and 3 as well. By compiler, I mean GCC or armclang and the version like armc6.15 or GCC 7..x.y . Reason I ask it is there can be issues like https://github.com/tensorflow/tensorflow/issues/37361 where compiler and its option matter.

felix-johnny, 2021-09-29:
@11223uhh I am closing this ticket for insufficient info. If you get some more info , please re-open.

closed 2021-09-29 by felix-johnny