ARM-software/CMSIS_5 #1270: RTX 5 pre-built libraries missing

KarolisMil, 2021-08-03:
Since newer release I have noticed, that RTX5 pre-built libraries are no longer present in the release pack.
1. Can they be got by some other way than building locally?
2. Is there any official documentation how to build it locally? I can see some scripts, but their usage is not documented anywhere I can find.

JonatanAntoni, 2021-08-03:
Hi @KarolisMil,

The RTX5 libraries are still be there in the pack found at https://github.com/ARM-software/CMSIS_5/releases/download/5.8.0/ARM.CMSIS.5.8.0.pack. We only removed DSP libraries in 5.8.0.

The folder CMSIS/RTOS2/RTX/Library/ contains a subfolder per target compiler. Underneath you can find the projects that are used to build libraries from source. Using MDK or IAR EW you should be able to open and build these projects locally, if required.

The libraries for Arm Compiler and GCC can be built from command line by using the provided `build.py` script. This script requires the Python module [python-matrix-runner](https://pypi.org/project/python-matrix-runner/) to be installed. Its command line is self-documented. The script itself just triggers the headless build of MDK, hence MDK needs to be installed and provided in system path.

Cheers,
Jonatan

tobermory, 2021-11-09:
On a sort-of-related note, in the 5.6.0 release, most sources of RTX5 could be built as a standalone .a file.  Only RTX_Config.h and rtx_lib.c needed to be copied to an application and configured to suit that application. I use GCC. 

It appears that in the 5.8.0 release, for CM3/4 at least, Source/GCC/irq_armv7m.S and Config/RTX_Config.c must be added to that list, since they either include (new) rtx_def.h, or rtx_os.h, both which in turn include RTX_Config.h.

It appears to me that a pre-built library is becoming harder to achieve in 5.8.0, compared to 5.6.0 at least.  Am I fighting a losing battle trying to partition RTX5 into these two 'components', a .a and a small .c+.S set that must be built per application?

RobertRostohar, 2021-11-09:
It should be still easy to have a pre-built library and adding only the rtx_lib.c with RTX_Config.h and RTX_Config.c (this was always the case).

It is true that the GCC/irq_*.S modules include rtx_def.h which also includes RTX_Config.h. However that is only used to determine if "Stack overrun checking" (OS_STACK_CHECK) is enabled. 

Therefore the irq_* modules can be part of the pre-built library. It is only not possible to change the OS_STACK_CHECK through RTX_Config.h anymore in the final image since this is fixed when creating the pre-built library. OS_STACK_CHECK requires RTX source variant (also stated in the RTX_Config.h).

This is exactly how the shipped pre-built libraries are created (with OS_STACK_CHECK=0).