ARM-software/CMSIS_5 #1274: The result of the arm_mat_solve_lower_triangular_xxx is wrong.

windwolf, 2021-08-06:
LT=[1.0016 0; 0.9984 0.5773], A=[1; 1]. 
expected DST=[0.9984; 0.0057], but DST=[0.9984; 0.9984].

I checked the source code:
``` c
arm_status arm_mat_solve_lower_triangular_f32(
  const arm_matrix_instance_f32 * lt,
  const arm_matrix_instance_f32 * a,
  arm_matrix_instance_f32 * dst)
  {
  arm_status status;                             /* status of matrix inverse */

#ifdef ARM_MATH_MATRIX_CHECK
  /* Check for matrix mismatch condition */
  if ((lt->numRows != lt->numCols) ||
      (a->numRows != a->numCols) ||
      (lt->numRows != a->numRows)   )
  {
    /* Set status as ARM_MATH_SIZE_MISMATCH */
    status = ARM_MATH_SIZE_MISMATCH;
  }
  else

#endif /* #ifdef ARM_MATH_MATRIX_CHECK */

  {
    /* a1 b1 c1   x1 = a1
          b2 c2   x2   a2
             c3   x3   a3

    x3 = a3 / c3 
    x2 = (a2 - c2 x3) / b2

    */
    int i,j,k,n;

    float32_t *pX = dst->pData;
    float32_t *pLT = lt->pData;
    float32_t *pA = a->pData;

    float32_t *lt_row;
    float32_t *a_col;

    n = dst->numRows;

    for(j=0; j < n; j ++)
    {
       a_col = &pA[j];

       for(i=0; i < n ; i++)
       {
            float32_t tmp=a_col[i * n];

            lt_row = &pLT[n*i];
            
            for(k=0; k < i; k++)
            {
                tmp -= lt_row[k] * pX[n*k+j];
            }

            if (lt_row[i]==0.0f)
            {
              return(ARM_MATH_SINGULAR);
            }
            tmp = tmp / lt_row[i];
            pX[i*n+j] = tmp;
       }

    }
    status = ARM_MATH_SUCCESS;

  }

  /* Return to application */
  return (status);
}
```
Then i found that the algorithm has the following constraints on the size of the matrix:

- lt->numRows == lt->numCols
- a->numRows == a->numCols
- lt->numRows == a->numRows

So my A does not satisfy the second rule.

---

I don't understand why a must be a square matrix, so I will continue to look at the code. The original intention of the algorithm should be to allow the number of columns of dst and a to be greater than 1, so that multiple linear equations can be solved at once, but there is a problem with the code processing. It assumes that a and dst are square matrices (the number of columns and the number of rows share the same Variable n), this assumption is completely unnecessary, maybe a->numCols == dst->numCols is enough.

The modified code flagments are as follows:
``` c
...

  if ((lt->numRows != lt->numCols) ||
      (lt->numCols != dst->numRows) ||
      (dst->numCols != a->numCols) ||
      (lt->numRows != a->numRows))

...

    int i,j,k,n,m; // add m

    n = dst->numRows;
    m = dst->numCols; // init m

    float32_t *pX = dst->pData;
    float32_t *pLT = lt->pData;
    float32_t *pA = a->pData;

    float32_t *lt_row;
    float32_t *a_col;

    for(j=0; j < m; j ++) // replace n with m
    {
       a_col = &pA[j];

       for(i=0; i < n ; i++)
       {
            lt_row = &pLT[n*i];

            float32_t tmp=a_col[i * m]; // replace n with m
            
            for(k=0; k < i; k++)
            {
                tmp -= lt_row[k] * pX[m*k+j]; // replace n with m
            }

            if (lt_row[i]==0.0f)
            {
              return(ARM_MATH_SINGULAR);
            }
            tmp = tmp / lt_row[i];
            pX[i*m+j] = tmp; // replace n with m
       }

    }

...
```

---
All the functions of arm_mat_solve_XXXXX_trianglar_XXX has the same problem.

christophe0606, 2021-08-06:
@windwolf Thank for the report.
Initially those functions were written in the context of an algorithm generating square matrix. But I agree that they should work with more general matrices.

I am leaving on vacations so I won't be able to look at it now. But I'll do when I am back.

christophe0606, 2021-08-31:
@windwolf It should be corrected in commit cbfe6e37b2d1282f754edc6af4d22bb1073dc4b2

Indeed, there were no good reasons for this restriction of the algorithm and now it can work with A matrix of any size.

Thanks for reporting it.

JonatanAntoni, 2021-09-06:
@windwolf, please consider closing this issue if your request is solved.

windwolf, 2021-09-28:
Great! Thanks all!

closed 2021-09-28 by windwolf