ARM-software/CMSIS_5 #1275: Omit unused samples from RFFT/RIFFT twiddle factor tables

richardallenatgarmin, 2021-08-06:
It's possible to reduce the size of realCoefA, realCoefB, realCoefAQ31, realCoefBQ31, realCoefAQ15,realCoefBQ15 when not using the 8192 point FFT by reducing the twidCoefRModifier values accordingly , which both reduces the data space used and can give a small performance benefit when more densely packing the arrays.

I'd like to suggest that we split the tables in the same way that twiddleCoef_rfft_NFFT has done, but build in only a single, largest used realCoefA_NFFT and adjust twidCoefRModifier accordingly.

In this way, still only a single table will be linked in, but now reflecting the size required for the largest supported FFT, and smaller FFTs can share the largest table in the way they do today.

For my application, this saves around 15kB of Flash since we're only using several small FFTs.

christophe0606, 2021-08-31:
@richardallenatgarmin I agree. Those big tables are not good. We have made a big effort with other tables to enable a better control of the code size through lots of compiler defines.
This work must continue with those remaining big tables.

I can't communicate a schedule but at some point it will be done.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606