ARM-software/CMSIS_5 #1277: Different results in arm_convolve_HWC_q15_fast and arm_convolve_HWC_q15_basic function

newton1010, 2021-08-10:
I try to use q15 convolution functions. In order to get shorter runtime, I use  "arm_convolve_HWC_q15_fast " function instead.
But I got different results in "arm_convolve_HWC_q15_basic" and "arm_convolve_HWC_q15_fast " functions with same input parameters.

Then, I verify the results without  ARM_MATH_DSP. The "arm_convolve_HWC_q15_fast" result is incorrect.
How can I fix it?

felix-johnny, 2021-08-13:
@newton1010  A few questions on this..
1. Are ch_im_in and ch_im_out are multiple of 2 ?
2. If possible, could you share a representative dimensions that you are using for the API? 
As for how to to fix it, printing out the sum values(before SSAT) and comparing is an option.

 I would like to point out this link about the different API's in use  https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#legacy-vs-tfl-micro-compliant-apis . We are in the process of adding optimizations for int16 input data and int8 weights now with TFLM quantization scheme. Maybe that is something that is of interest to you.

felix-johnny, 2021-08-13:
Issue to be followed in https://github.com/ARM-software/CMSIS_5/issues/1262

closed 2021-08-13 by felix-johnny