ARM-software/CMSIS_5 #1279: Finding the phase of a specific harmonic with arm_cfft_F32()

FlixofTheWrist, 2021-08-11:
Hey, I have a question on the same subject: calculating the phase of a chosen frequency of the arm_cfft_f32() function. I will start by telling you about my application and code.

The input buffer is 8192 samples of a 60Hz perfect sin signal. (for each real number, I associate a 0)
The sample speed is 20 480 Hz.
Since sampling speed is 20 480Hz and buffer of the output is 4096, my frequency precision is 5Hz. (5*12 = 60Hz)
So in the Input buffer, after computing the CFFT, bin of fundamental should be 24 and 25. (60Hz real and imaginary).
Input analog signal is generated by Analog Discovery 2 and is a sin wave that is centered at 1.67V with amplitude of 1.67V.
The phase of the analog sin wave is set at 0 on the software provided (waveform) by the Analog Discovery 2.

I am building a code that can find the harmonics on a 60Hz experimental power grid.
The function arm_cmplx_mag_f32() works just fine to find the magnitude of detected harmonics. Good precision.
The problem is with the phase. It is important to know the phase to be able to apply the right correction on the given harmonic to eliminate it completly.

So, i tried using the atanf2 function to find the phase of a given real and imaginary pair.

However, the problem is that after computing the CFFT multiple times, the value are never the same in the INPUT buffer that was affected by CFFT for the SAME sin wave.

In other words, atanf2 function works just fine probably, it's about the return values of the CFFT.

Do you guys know why i can't calculate the phase with precision but I can find the magnitude easily.

christophe0606, 2021-08-30:
@FlixofTheWrist 

> However, the problem is that after computing the CFFT multiple times, the value are never the same in the INPUT buffer that was affected by CFFT for the SAME sin wave.

I am not sure I understand:

- Do you mean the FFT result is different with the same input buffer ? 
- Or the FFT result is different with the same digital sin wave but possibly sampled at different time offset (and so with different input buffer) ?
- Or the same analog sin is used so the digital one may be different from one run to another (small noise) ?

Is it the FFT result which is the problem ? Or it is the phase ? 

For the phase, have you tried to display it ? Is it a problem of unwrapping of the phase (ambiguity of +- 2Pi) or something else ?

FlixofTheWrist, 2021-08-31:
Thanks for your awnser @christophe0606! I will try to explain it more.

The FFT result is different with the same digital sin wave. The 60 Hz sinwave is generated using ''Analog Discovery 2''. The 60 Hz sinwave is the input of the ADC of the STM32F767ZI and is always running. So, i'm doing FFTs on the same signal, yet the value of the outputs of the FFT is never the same (for the same generated 60Hz sinwave). The problem is the FFT result, since it is never the same, i can't use the values to calculate the phase of a given harmonic. Since i'm doing the FFT on a ''perfect'' 60Hz sinwave that is programmed on the application of the generator to have a phase of 0, im supposed to get a phase of 0. Yet, i get different results and the results are never the same. I calculate it with atanf2 function. Is this explanation more precise?

Thanks

Félix

christophe0606, 2021-09-01:
@FlixofTheWrist Thanks for additional details.

The FFT is modifying the input buffer so perhaps it may be the issue if you was not aware of this.

Otherwise there is no reason internal to the FFT implementation that could explain that the FFT is giving different results for the same FFT input buffer.

So it may be due to an external corruption : some of the tables or buffers used by the FFT ?

Otherwise for the phase, except the problem of phase wrapping atanf2 should be ok.

Do you have a reproducer ? An input buffer (or a few input buffers) that could be used to test or reproduce the problem on my side ?

FlixofTheWrist, 2021-09-01:
@christophe0606 Thank you for your response, this is really appreciated!

I will comeback with another response with the infomation you want. This will make it easier to find the problem.

Thanks again,

Félix

closed 2021-10-04 by christophe0606