ARM-software/CMSIS_5 #1285: arm_mve_tables_f16.c, "floating-point value does not fit in required floating-point type"

TTornblom, 2021-08-16:
While working on an IAR compiler update, it was noted that the file "arm_mve_tables_f16.c" contains floating point initializers that does not fit into the float16 data type.

`Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 59 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 112 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 122 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 279 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 319 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 329 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 902 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 1062 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 1102 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 1112 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 3349 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 3989 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 4149 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 4189 
 Error[Pe221]: floating-point value does not fit in required floating-point type D:\CMSIS-5.8\cmsis-5.8-nid\CMSIS\DSP\Source\CommonTables\arm_mve_tables_f16.c 4199 
`

I assume it is the value "(float16_t)0.00000000000000006123f" that does not fit into a float16 and gets rounded to 0.

This seems like a bug to me.

christophe0606, 2021-08-30:
@TTornblom Thanks for reporting it. It is not a bug but may be done in a cleaner way.
The values are computed and generated by a Python script. The computation is in double precision.

The generated file is containing a cast to (float16_t) which could (with some compilation options) generate a warning since it is decreasing the accuracy. But it is normal.

Another possibilities would be to generate from Python with the right accuracy and avoid this cast.

I'll add it as enhancement if you confirm it is not preventing you from building.

TTornblom, 2021-08-30:
Thanks,
This is not preventing us from building, we can disable this error and we will likely change it into a warning.

TTornblom, 2021-10-01:
Looks like the developers would like to get the sources fixed instead of changing the compiler behavior. Can we get this done?

christophe0606, 2021-10-04:
@TTornblom It can be done. I'll try to prioritize it.

TTornblom, 2021-10-04:
Thanks, much appreciated.

christophe0606, 2021-10-04:
@TTornblom It should be corrected in commit https://github.com/ARM-software/CMSIS_5/commit/156cc72c04607c1eacf0f90159b62dfe6dfee5ea

Can you confirm ? Because I am unable to get the issue with gcc and clang.

According to this quote from the C standard:

> The translation-time conversion of floating constants should match the execution-time
> conversion of character strings by library functions, such as strtod, given matching
> inputs suitable for both conversions, the same result format, and default execution-time
> rounding.

it may be normal that I am not getting any warning with gcc and clang (and I have tested with several warning options).

But the sentence 

> given matching inputs suitable for both conversions

may be understood as saying I should get warnings after all.

Anyway, the change was done.

TTornblom, 2021-10-04:
Yes, this seems to fix this issue.

Thanks!

closed 2021-10-04 by christophe0606