ARM-software/CMSIS_5 #1288: CMSIS SVD validation failed for Cortex-M33

radhika-raghavendran, 2021-08-19:
When running the schema validator in \CMSIS\Utilities\CMSIS-SVD.xsd against the SVD file for our microcontroller, we find some errors. According to the documentation in https://arm-software.github.io/CMSIS_5/SVD/html/elem_registers.html#elem_register and https://arm-software.github.io/CMSIS_5/SVD/html/elem_peripherals.html#elem_peripheral it is allowed in the SVD.
Can you check if the schema file requires an update?

Attached are the errors in the SVD file which running the schema validation tool.
[image]
[image]
[image]
[image]

JonatanAntoni, 2021-09-06:
Hi @radhika-raghavendran,

Looking into the schema definition turns out that all these elements are declared to be in a sequence, i.e. need to be in order. Please try to arrange the elements according to the order in the schema and check again. I expect this to work.

I am going to clarify if there is a (technical) requirement for this restriction.

Cheers,
Jonatan

radhika-raghavendran, 2021-09-13:
Thank you for the clarification. However there is still an issue with the validation with schema given in the ARM_CMSIS distribution.
According to the specifications, there is no requirement to include a description in the cluster element, as shown below.
[image]
I get a validation error when I do not include the description tag in the cluster element, as shown here:
[image]
When I include the description tag, with no other changes, the validation is complete and successful as shown here.
[image]

JonatanAntoni, 2021-09-13:
Hi @radhika-raghavendran,

yes, you are right. We need to clarify the documentation.
The current schema mandates a description in this case.

Cheers,
Jonatan

radhika-raghavendran, 2021-09-14:
The schema validation is now complete. Please add required sections and make modifications to your documentation.
I will close this ticket.

radhika-raghavendran, 2021-09-14:
Closed ticket. Documentation for SVD file to be modified by the ARM team.

closed 2021-09-14 by radhika-raghavendran