ARM-software/CMSIS_5 #1291: DSP: Buffer overflow in test_mat_vec_mult_*

stephanosio, 2021-08-20:
There is a buffer overflow in the `test_mat_vec_mult_*` tests.

https://github.com/ARM-software/CMSIS_5/blob/13b9f72f212688d2306d0d085d87cbb4bf9e5d3f/CMSIS/DSP/Testing/Source/Tests/UnaryTestsQ15.cpp#L127

The maximum value of `internal` is 23.

In `PREPAREVECDATA2`, it can copy up to `23 * 2 = 46` elements to `bp` (= `b.ptr()`).
https://github.com/ARM-software/CMSIS_5/blob/13b9f72f212688d2306d0d085d87cbb4bf9e5d3f/CMSIS/DSP/Testing/Source/Tests/UnaryTestsQ15.cpp#L115

For `TEST_MAT_VEC_MULT_Q15_5`, the max. number of elements for `b` is `MAXMATRIXDIM`.
https://github.com/ARM-software/CMSIS_5/blob/13b9f72f212688d2306d0d085d87cbb4bf9e5d3f/CMSIS/DSP/Testing/Source/Tests/UnaryTestsQ15.cpp#L341

Note that `MAXMATRIXDIM` is 40 and this is less than 46 elements copied to `b.ptr()`:
https://github.com/ARM-software/CMSIS_5/blob/13b9f72f212688d2306d0d085d87cbb4bf9e5d3f/CMSIS/DSP/Testing/Source/Tests/UnaryTestsQ15.cpp#L19

christophe0606, 2021-08-31:
It should be corrected in commit 6e513eab300bdc3eea6d2c1df1af7cdb14868055
Some part of the test was wrongly assuming complex matrixes and thus using an additional 2 factor.

I have also updated the other mat / vec multiply tests.

stephanosio, 2021-08-31:
> It should be corrected in commit [6e513ea](https://github.com/ARM-software/CMSIS_5/commit/6e513eab300bdc3eea6d2c1df1af7cdb14868055)
> Some part of the test was wrongly assuming complex matrixes and thus using an additional 2 factor.
> 
> I have also updated the other mat / vec multiply tests.

Thanks! Closing since this issue has been resolved.

closed 2021-08-31 by stephanosio