ARM-software/CMSIS_5 #1293: DSP: Missing const qualifier in arm_linear_interp

stephanosio, 2021-08-22:
The `arm_linear_interp` functions are missing `const` qualifier for the interpolation table argument (`pYData`).

https://github.com/ARM-software/CMSIS_5/blob/13b9f72f212688d2306d0d085d87cbb4bf9e5d3f/CMSIS/DSP/Source/InterpolationFunctions/arm_linear_interp_q7.c#L54

christophe0606, 2021-08-31:
It should be corrected in commit 469840909e8190511d17e4953958c4e9234254dc.
I have also updated the other functions (q15, q31 )

stephanosio, 2021-08-31:
> It should be corrected in commit [4698409](https://github.com/ARM-software/CMSIS_5/commit/469840909e8190511d17e4953958c4e9234254dc).
> I have also updated the other functions (q15, q31 )

Thanks! Closing since this issue has been resolved.

closed 2021-08-31 by stephanosio