ARM-software/CMSIS_5 #1305: The definition of Cosine Distance isn't quite clear

jbr-smtg, 2021-09-09:
Hello :)

I'm not quite sure to understand what to expect from the Cosine Distance functions arm_cosine_distance.
Initially I understood that it could be the cosine similarity (https://en.wikipedia.org/wiki/Cosine_similarity) but it seems to provide results which are different.

A little clarification/documentation about what was expected to be the cosine distance in that case could help.

Many thanks!

christophe0606, 2021-09-10:
@jbr-smtg The cosine_distance(u,v) is 

`1 - u . v / (Norm(u) Norm(v)) `

in CMSIS-DSP. I'll add it to the documentation for the function.

`u . v / (Norm(u) Norm(v)) `

is the cosine similarity

christophe0606, 2021-09-10:
Corrected in commit 285816e38ec8cc3289acefc1786a28726be0929d
The definition was added to the doxygen documentation

jbr-smtg, 2021-09-10:
My bad, it indeed totally makes sense. Thanks for the clarification!

closed 2021-09-30 by christophe0606