ARM-software/CMSIS_5 #1309: CMSIS-NN: Calculate quant_params in arm_convolve_wrapper_s8

MiguelCosta94, 2021-09-16:
Hi,

I am trying to deploy a model trained in Tensorflow to an ARM Cortex-M MCU, using CMSIS-NN. Since I want to keep the overhead as low as possible I am not using the TFLite C libraries. I pretend to use the CMSIS-NN library directly. I am able to perform a per-layer quantization and use the q7 and q15 functions successfully, but I would like to compare the performance and accuracy of the model using the s8 functions. However, I can not find any documentation for the parameters ctx and quant_params of arm_convolve_wrapper_s8. Can you provide more information about these parameters and how to calculate them?

felix-johnny, 2021-09-16:
@MiguelCosta94 Thanks for the question. Let me answer your question first before posting some questions about what you are trying to do. Please refer to ticket https://github.com/ARM-software/CMSIS_5/issues/1259 which answers the question on how to calculate the quantization parameters. The tflite:: function specified there is from https://github.com/tensorflow/tflite-micro 

 As for ctx, it is a means for the CMSIS-NN API to inform the user if a scratch buffer is required for processing like for e.g im2col buffers. https://github.com/tensorflow/tflite-micro/blob/fc6fede6c89e103b60a840348ae9c3866d5f507d/tensorflow/lite/micro/kernels/cmsis_nn/conv.cc#L115 and https://github.com/tensorflow/tflite-micro/blob/fc6fede6c89e103b60a840348ae9c3866d5f507d/tensorflow/lite/micro/kernels/cmsis_nn/conv.cc#L203 is an example of how to use it.

  The link https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#legacy-vs-tfl-micro-compliant-apis gives you the difference between the s7 and s8 functions and the recommended and supported API's are the _s8() ones. With that in mind, you could convert your Tensorflow model into a quantized one using information in https://www.tensorflow.org/lite/performance/post_training_quantization and then use the functions mentioned in #1259 to obtain the multiplier and scaling values(if you do not want to use the TFLM interpreter). 

 Hope that helps.

felix-johnny, 2021-09-17:
@MiguelCosta94 An additional comment on the overhead that you mention. The cycle overhead is quite minimal for the popular models out there. It can be as low as a fraction of a single percentage. The memory required TFLM runtime is in the sub 20 kB range. Using the runtime can save you development and maintenance time. Hope this helps in your design choices.

felix-johnny, 2021-09-29:
@MiguelCosta94  I am closing this ticket as there is no further activity. If you have any further questions don't hesitate to reach out!
Thanks..

closed 2021-09-29 by felix-johnny