ARM-software/CMSIS_5 #1312: API enhancements

ml-0, 2021-09-20:
Hi,

are there API functions in CMSIS DSP for

- fill f64: to be used to initialize a complex f32 array with constant values, similar to arm_fill_f32 for real values?
- extract the real/imaginary parts of a complex array into a real array?
- sum up all elements of a real vector into a scalar value?

Maybe I missed that one of the above functionality can be mapped to an existing function in the CMSIS DSP API?

Best regards
ml-0

christophe0606, 2021-09-20:
@ml-0 arm_fill_f64 was added very recently in the develop branch so it is now available.

You're right that we do not have the other two functions and with vectorization (M55/Neon) it would be interesting to have them.

christophe0606, 2021-09-20:
@ml-0 Hi,

Another question : For the accumulation, which datatype would interest you more ? f32 ? f16 ? q31 ? q15 ?

And if accumulating fixed points: what should be the format of the output ? For instance : accumulating a q31 into a q64 ? Accumulating a q15 into a q31 ?

Thanks.

ml-0, 2021-09-20:
Currently, f32. Consequently, I did not think of the output format of the element sum in q31/q15. In my use case I do not even use the absolute value. I just use the result in comparisons of several element sums to find the largest one.

So, if I did the quantized implementation, the best fit would be accumulating q31 into q63 because I am heading for speed, avoiding any scaling/division of the result.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606