ARM-software/CMSIS_5 #1358: Make LOW_OPTIMIZATION_ENTER/EXIT switchable or consider removing it

kalj, 2021-11-02:
We use the cascaded_biquads_df2T_f32 on Cortex M4f (a BES2300Y chip specifically). We don't get decent performance unless we remove the LOW_OPTIMIZATION_ENTER/EXIT macros, which seem unnecessary. We have not seen any problem with removing them.

The question is what the point of these macros are? What to they guard against?

At least we would really appreciate if it was possible to switch this behavior off with some macro.

christophe0606, 2021-11-02:
@kalj I think those macros are legacy and no more really useful.
I assume they were introduced in the past as workarounds for some compiler issues.
I'll remove them from the code.

christophe0606, 2021-11-16:
Corrected in commit 6368a853f29d641ad7a7759e17bee557ac26d8cd

closed 2021-11-16 by christophe0606