ARM-software/CMSIS_5 #1368: Questions about CMSIS-NN_INTQ

mengna0707, 2021-11-16:
May I ask if anyone is using CMSIS-NN-INTQ for deployment?

Is there anyone who can answer some questions about the conversion of parameters and the precision Settings involved in using XNOR-Net?

felix-johnny, 2021-11-16:
@mengna0707 CMSIS-NN-INTQ  is a fork from CMSIS-NN that is not maintained by Arm. Unfortunately, I don't have an answer to your question. 

 But, I can direct you to a blog post on how to get started with CMSIS-NN https://community.arm.com/arm-community-blogs/b/ai-and-ml-blog/posts/image-recognition-on-arm-powered-microcontrollers 
 Hope that helps.

mengna0707, 2021-11-18:
Hi @felix-johnny,thank you very much for your reply and suggestions.  
Unfortunately, most of my recent work has been in binary networks. 
But thanks again for your reply.

closed 2021-11-18 by felix-johnny