ARM-software/CMSIS_5 #1371: Cortex M7 RTOS2 (RTX5) library

KarolisMil, 2021-11-18:
Hello,
I have looked into RTX5 pre-built libraries and cannot find one for Cortex M7. Can you please advise, why it is not there and what to use with Cortex M7?
Thank you in advance.

JonatanAntoni, 2021-11-18:
Hi @KarolisMil,

The Cortex-M7 is sharing same architecture with Cortex-M4, both are Armv7E-M. The M7 has caches which the M4 one does not have. But this difference is transparent to RTX5, i.e., RTX5 doesn't need to know about the caches. Hence we are using the same libraries for both.

You can find this information by inspecting the PDSC file, see
https://github.com/ARM-software/CMSIS_5/blob/72cdcc54902bf702f79c91e8ab172e28ac568f3e/ARM.CMSIS.pdsc#L3057-L3060

The only relevant feature for RTX5 is FPU. If your M4/M7 does not include/use an FPU, the device is seen similar to an M3. With FPU the kernel needs to consider the extended stack frame format including FPU registers.

Let me know if this anwers your question.

Cheers,
Jonatan

KarolisMil, 2021-11-18:
Thank you very much for your explanation.