ARM-software/CMSIS_5 #1373: Cannot Build NN module on host

EdXian, 2021-11-22:
Hi all,
I have found that some macro is not defined when building the NN module with CMSIS-DSP.

target: Host (Ubuntu)
version: the latest develop
build command: `cmake -DHOST=ON ..`
[image]

the related issue:
https://github.com/ARM-software/CMSIS_5/issues/1149
https://github.com/ARM-software/CMSIS_5/issues/1015

mansnils, 2021-11-25:
Hi @EdXian 
CMSIS-NN can be built independently and that is the recommended way. Here are some instructions: https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/README.md
Hope that helps.

EdXian, 2021-11-25:
Hi @mansnils ,
Thank you for your reply. 
I think my question is : I want to build to NN package not only for embedded system but for linux/mac system. I found that some of nn support function is based on  the arm assembly machine code which cannot be compiled by host. However, Accroding to https://github.com/ARM-software/CMSIS_5/issues/1015, the nn module can be built by configuring cmake flag `-DHOST=on`.
Any idea?

christophe0606, 2021-11-25:
@EdXian The flag -DHOST=on is a feature of the CMSIS-DSP cmake.
CMSIS-NN is built differently and recognize different options.

EdXian, 2021-11-25:
Hi @christophe0606,
You are right. I  checked all the flag for NN package and there is no flag or macro called `Host` .
Is there a way to compile the NN module for linux/Mac system?

EdXian, 2021-11-25:
@christophe0606 
For some reasons, I have to verify the input/output of the neural network in different system.
Is it recommended to use Tensorflow Lite C API to verify the CMSIS-NN module?

christophe0606, 2021-11-26:
@EdXian I am maintaining CMSIS-DSP but not NN. So, I'll let NN team answer.

EdXian, 2021-11-26:
@christophe0606 ,
Thank you very much

mansnils, 2021-11-26:
CMSIS-NN comes in three flavors, pure C, and optimizations for DSP and MVEI. The optimizations flag are automatically enabled give the corresponding Cortex-M so the default is the pure C version.
So although I haven't tried it myself it should be possible to build CMSIS-NN for host, with for example a toolchain for Linux.

It is recommended to use CMSIS-NN with Tensorflow Lite Micro. Here is a starting point: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/cmsis_nn/README.md

EdXian, 2021-12-02:
Hi @mansnils ,
well, I think some functions are not included into the CMakelist.txt in NN module. It will cause the compiler error. 

For example, the function `arm_nn_add_q7` is not included in CMakeLists.txt

https://github.com/ARM-software/CMSIS_5/blob/1fab31865cd538f12d1f74b760939404771603e3/CMSIS/NN/Source/NNSupportFunctions/CMakeLists.txt#L20-L25

Should I create a new issue for it?

mansnils, 2021-12-02:
Hi @EdXian,
arm_nn_add_q7 is part of the legacy API: https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/README.md#legacy-vs-tfl-micro-compliant-apis
We just build the TFLM compliant API. Would it be ok for you to just build that?

EdXian, 2021-12-03:
Hi @mansnils ,
Now , It's ok for me to build the source. 
Thank you very much!

closed 2021-12-03 by EdXian

EdXian, 2021-12-08:
Hi @mansnils ,
I want to try [NNoM](https://github.com/majianjia/nnom) which only supports legacy CMSIS-NN API. Is it possible to build legacy NN API by modifying some flags in cmake? I find that some basic function (arm_relu_q7 ...) are not included to CMakeList.txt.

mansnils, 2021-12-08:
Hi @EdXian 
Sure it should be possible. As mentioned the legacy API is not supported but contributions are always welcome. If you want to contribute I think it should be under a flag.