ARM-software/CMSIS_5 #1382: [BUG] `r1` overriden in SVC, GCC ignores it when compiling with LTO and Ofast

RisinT96, 2021-12-19:
Encountered the bug when trying to allocate two semaphores with the same arguments.
 
Inside the SVC multiple operations are performed:
* Finds the stack pointer of the calling thread
* Extracts the stacked registers {r0-r3,r12}
* Calls the service function
* Stores {r0, r1} into the stacked registers of the calling thread
 
Once the SVC returns, the stacked registers are loaded into the cpu.
 
Thus r0 and r1 have been overridden, in the case of allocating semaphores, r0 contains the semaphore address and r1 appears to contain 0.
 
When compiling with LTO, in some specific cases GCC seems to ignore the fact that r1 has changed and loads it back into r0 before the second SVC call, resulting in both arguments being 0 and the mutex allocation to fail.
 
I created the simplest example I could to demonstrate the bug, had to create my own SVC, as it happens only when the semaphore allocation is inlined by LTO, which is hard to enforce in a tiny project. 
I simply expanded the macros that make up `SVC0_3(SemaphoreNew,      osSemaphoreId_t, uint32_t, uint32_t, const osSemaphoreAttr_t *)` and removed the attributes argument as it seems to bear no effect.
 
Here is my example code that demonstrates the bug:
```c
#include <stdbool.h>
#include <stddef.h>
#include <stdint.h>
 
static void *rtos_call_svc(uint32_t max_count, uint32_t initial_count)
{
    return (void *)0xDEADBEEF;
}
 
void *rtos_call(uint32_t a1, uint32_t a2)
{
    register uint32_t __r0 __asm("r0") = (uint32_t)a1;
    register uint32_t __r1 __asm("r1") = (uint32_t)a2;
    register uint32_t __rf __asm("r12") = (uint32_t)rtos_call_svc;
    /* Notice how `r1` is not mentioned in the `OutputOperands` nor the `Clobbers` */
    __asm volatile("svc 0"
                   : "=r"(__r0)
                   : "r"(__rf), "r"(__r0), "r"(__r1)
                   :);
    return (void *)__r0;
}
 
int main(void)
{
    // Doesn't work:
    // * r1 and r0 are overriden inside SVC
    // * r1 is loaded into r0
    // * the second call is executed with incorrect arguments
    void *res_1 = rtos_call(0x31337, 0x31337);
 
    if (res_1 == NULL)
    {
        for (;;)
        {
        }
    }
 
    rtos_call(0x31337, 0x31337);
 
    // Doesn't work:
    // * r1 and r0 are overriden inside SVC
    // * the second call is executed with incorrect arguments
    rtos_call(0xDEADC0DE, 0xDEADC0DE);
    rtos_call(0xDEADC0DE, 0xDEADC0DE);
 
    // Works:
    // * r1 and r0 are overriden inside SVC
    // * r0 is loaded with `0xB00B1E5`
    // * r0 is moved into r1
    // * the second call is executed with correct arguments
    void *res_2 = rtos_call(0xB00B1E5, 0xB00B1E5);
 
    if (res_2 == NULL)
    {
        __asm volatile("nop");
    }
 
    rtos_call(0xB00B1E5, 0xB00B1E5);
}
```
 
Which results in the following assembly being generated by gcc version 8.3.1 20190703 (release) [gcc-8-branch revision 273027] (GNU Tools for Arm Embedded Processors 8-2019-q3-update) (also tested with gcc 10.3-2021.10 with equal results):
```assembly
main
$Thumb
; First variation - broken
 08001154   LDR            R0, =0x00031337               ; [PC, #48] [0x08001188]               ; Load 0x31337 into R0
 08001156   LDR.W          R12, =rtos_call_svc           ; [PC, #60] [0x08001194] =0x08000511
 0800115A   MOV            R1, R0                                                               ; Load R0 into R1
 0800115C   SVC            #0                                                                   ; R0 and R1 are overriden inside the SVC
 0800115E   CBZ            R0, 0x08001186                ; <main>+0x32
 08001160   MOV            R0, R1                                                               ; Copy overriden R1 into R0, both arguments now invalid
 08001162   SVC            #0                                                                   ; Call SVC with incorrect arguments
; Second variation - broken
 08001164   LDR            R0, =0xDEADC0DE               ; [PC, #36] [0x0800118C]               ; Load 0xDEADCODE into R0
 08001166   MOV            R1, R0                                                               ; Load R0 into R1
 08001168   SVC            #0                                                                   ; R0 and R1 are overriden inside the SVC
 0800116A   MOV            R0, R1                                                               ; Copy overriden R1 into R0, both arguments now invalid
 0800116C   SVC            #0                                                                   ; Call SVC with incorrect arguments
; Third variation - works
 0800116E   LDR            R0, =0x0B00B1E5               ; [PC, #32] [0x08001190]               ; Load 0xB00B1E5 into R0
 08001170   MOV            R1, R0                                                               ; Load R0 into R1
 08001172   SVC            #0                                                                   ; R0 and R1 are overriden inside the SVC
 08001174   CBNZ           R0, 0x08001178                ; <main>+0x24
 08001176   NOP
 08001178   LDR            R0, =0x0B00B1E5               ; [PC, #20] [0x08001190]               ; Correctly load 0xB00B1E5 into R0
 0800117A   LDR.W          R12, =rtos_call_svc           ; [PC, #24] [0x08001194] =0x08000511
 0800117E   MOV            R1, R0                                                               ; Load R0 into R1
 08001180   SVC            #0                                                                   ; Call SVC with correct arguments
 08001182   MOVS           R0, #0
 08001184   BX             LR
 08001186   B              0x08001186                    ; <main>+0x32
```
 
For a working/compiling example, please refer to my [repository](https://github.com/DickButt69WorldwideMkII/RTX-SVC-r0-r1-bug).
 
Looking at the following:
 
<table>
<tr>
<th>
`irq_armv7m.S`
</th>
<th>
`irq_armv6m.S`
</th>
</tr>
 
<tr>
<td>
 
```assembly
SVC_Handler:
        tst      lr,#0x04               // Determine return stack from EXC_RETURN bit 2
        ite      eq
        mrseq    r0,msp                 // Get MSP if return stack is MSP
        mrsne    r0,psp                 // Get PSP if return stack is PSP
 
        ldr      r1,[r0,#24]            // Load saved PC from stack
        ldrb     r1,[r1,#-2]            // Load SVC number
        cmp      r1,#0                  // Check SVC number
        bne      SVC_User               // Branch if not SVC 0
 
        push     {r0,lr}                // Save SP and EXC_RETURN
        ldm      r0,{r0-r3,r12}         // Load function parameters and address from stack
        blx      r12                    // Call service function
        pop      {r12,lr}               // Restore SP and EXC_RETURN
        stm      r12,{r0-r1}            // Store function return values
```
 
</td>
 
<td>
 
```assembly
SVC_Handler:
        mov      r0,lr
        lsrs     r0,r0,#3               // Determine return stack from EXC_RETURN bit 2
        bcc      SVC_MSP                // Branch if return stack is MSP
        mrs      r0,psp                 // Get PSP
 
SVC_Number:
        ldr      r1,[r0,#24]            // Load saved PC from stack
        subs     r1,r1,#2               // Point to SVC instruction
        ldrb     r1,[r1]                // Load SVC number
        cmp      r1,#0                  // Check SVC number
        bne      SVC_User               // Branch if not SVC 0
 
        push     {r0,lr}                // Save SP and EXC_RETURN
        ldmia    r0,{r0-r3}             // Load function parameters from stack
        blx      r7                     // Call service function
        pop      {r2,r3}                // Restore SP and EXC_RETURN
        stmia    r2!,{r0-r1}            // Store function return values
        mov      lr,r3                  // Set EXC_RETURN |
```
 
</td>
</tr>
 
</table>
 
r0 and r1 are always overridden before the service returns from the IRQ, thus I think they should both appear in either `OutputOperands` or  `Clobbers`.

RobertRostohar, 2021-12-20:
Thanks for the details analyses. Register "r1" needs to be either in the Output or Clobber list.

RobertRostohar, 2021-12-20:
Actually it is not necessary anymore that SVC handler stores register "r1". This was a left-over from the older API where also 64-bit values were returned. Now only 32-bit values are returned in register "r0".

RisinT96, 2021-12-20:
In that case we have two possible solutions:
* The one in my PR (though I accidentally forgot to update `SVC_Out1` to `SVC_Out2`, which I'll fix)
* Remove the override of `r1` in the `SVC_Handler`.

I can update my PR to reflect the solution of your choice.

RobertRostohar, 2021-12-20:
The problem should be fixed now in develop branch.

RisinT96, 2021-12-20:
LGTM

closed 2021-12-20 by RisinT96