ARM-software/CMSIS_5 #1386: arm_sqrt_q15 improvements

olikraus, 2022-01-03:
This issue is related to
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FastMathFunctions/arm_sqrt_q15.c

I started my own implementation of a q15 sqrt procedure here: https://github.com/olikraus/fast_sqrt .
During this work I found two possible enhancements.

## Upshift / Downshift of the input

Instead of checking whether the shift is odd or even (https://github.com/ARM-software/CMSIS_5/blob/3c79bde31a8f5f35b02436f7e45da02d9448c09d/CMSIS/DSP/Source/FastMathFunctions/arm_sqrt_q15.c#L65), the variable `signBits1` could be forced to be even.  This would avoid the if condition at the beginning and the end of the procedure.

https://github.com/olikraus/fast_sqrt/blob/bfc6933a11867b31296ae5ba588cf29872814499/fast_sqrt.c#L120-L123

## Precision 

```
    q15_t in;
    q15_t out = 0;
    q15_t out2;
    q15_t diff;
    for( in = 0; in != 0x7fff; in++ )
    {
      arm_sqrt_q15( in, &out);
      out2 = (((q31_t)out)*((q31_t)out))>>15;
      diff = in - out2;
      printf("%d: %d %d %d\n", in, out, out2, diff);
    }
```
Here is some output:
```
32019: 32394 32024 -5
32020: 32395 32026 -6
32021: 32396 32028 -7
32022: 32397 32030 -8
32023: 32398 32032 -9
32024: 32399 32034 -10
32025: 32400 32036 -11
32026: 32393 32022 4
32027: 32394 32024 3
32028: 32395 32026 2
32029: 32396 32028 1
```

This means the quadratic value of the result might differ by up to  +/-11 from the input value. As far as I understood the code, the root cause for this is the 4.12 internal fix point format for the 1/sqrt(n) approximation. Instead of 4.12 fix point, 2.14 seem to still deliver correct results while precision increases. Of course changing this requires also a change in the LUT. It may also require to change the internal variables to 32 bit (which would not have any impact on arm, because the registers are anyways 32 bit).

christophe0606, 2022-02-02:
@olikraus Thank you.
I tag this as enhancement and I'll come back to it much later when I have some bandwidth.

The sqrt is used in lots of other kernels so before changing it I'll of course have to check it is not changing the behavior of the other kernels. Also I'll need to check the scalar and vector version are giving similar results.

But, in the worse case, we may do with like some other functions and have a arm_sqrt_fast function : faster and a bit less accurate.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606