ARM-software/CMSIS_5 #1389: test the final reordering with a float version of CMSIS-NN

mengna0707, 2022-01-07:
I tried to use CMSIS-NN to build an inference model, which could run successfully, but the inference result was wrong.

I suspect two possible reasons：
1）An error occurred in the reordering of weights.
This tutorial（ https://developer.arm.com/documentation/102591/0000/Compare-the-ML-framework-and-CMSIS-NN-data-layouts ) suggests testing the floating point version of CMSIS-NN after sorting, but the reference implementation mentioned is not in CMSIS/NN.
Could you please tell me the test method ?

2）The calculation of bias shift and out shift is wrong.
I looked at previous issues, but I still have doubts about the Qm.n of input and activation:
How can I calculate activation Q if I do not use the official quantization method of Tensorflow?

Any suggestions would be greatly appreciated!

mansnils, 2022-01-07:
Hi @mengna0707 
It sounds like you used the Legacy API and not the TFL micro compliant API. If not please elaborate more what you have done.
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/README.md#legacy-vs-tfl-micro-compliant-apis

The tutorial you mentioned is based on the legacy API.
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/README.md#reference

Contributions are always welcome but the unfortunately the legacy API is not supported.

mengna0707, 2022-01-07:
Hi @mansnils ,Like you said, I'm using the Legacy API. Because I want to achieve less than 8-bit quantization, from what I've seen, TensorflowLite now does not support different quantization cases in a network (for example, 8-bit quantization for the first and last layers, 2-bit quantization for the middle layer). 

To implement the example above, I am now trying to quantize the 8-bit point of CMSIS_5( https://github.com/ARM-software/CMSIS_5 ) with CMSIS_NN-INTQ( https://github.com/EEESlab/CMSIS_NN-INTQ ) to implement the inference framework of the neural network.  

However, the process is difficult. So, if it's convenient, could you give me some guidance?I would appreciate it very much.

mansnils, 2022-01-10:
From https://github.com/ARM-software/CMSIS_5/issues/1390 it appears you solved this issue, so closing it.

closed 2022-01-10 by mansnils

felix-johnny, 2022-01-10:
@mengna0707 what is available in https://developer.arm.com/documentation/102591/0000/Choose-a-quantization-scheme is the only material available for support of the legacy APIs. For your first question on floating point, there are no floating point implementation in CMSIS-NN for the legacy or the TFLM compatible APIs. If you go the legacy API way, you'll have to create the floating point APIs' and any test framework attached to it.

  You could post a question on how one goes about experimenting with different quantization bits in TFLM in https://github.com/tensorflow/tflite-micro/issues

mengna0707, 2022-01-11:
Hi @felix-johnny ，thank you very much for your advice！
Regarding the testing issue, as #1390 said， I now have a preliminary approach and am currently testing the feasibility.
And  I will go to the TFLM community to look at the Settings and issues associated with quantification.