ARM-software/CMSIS_5 #1391: CMSIS-DSP: Documentation error arm_rfft_q31

tluliak, 2022-01-08:
https://arm-software.github.io/CMSIS_5/DSP/html/group__RealFFT.html#gabaeab5646aeea9844e6d42ca8c73fe3a

The documentation tables appear to indicate that the shift required to recover the input signal is one less than necessary if an RFFT is followed by RIFFT. The shift is equal to the power of two of the FFT length, thus the shift for an FFT with length 32 is 5, not 4 as indicated by the table. 

```
    for ( unsigned t = 0; t < 16; t++) 
    {
        if (FFT_Len >> t & 0x01) { scaleFac = t; }
    }
```
where `unsigned FFT_Len` is a power of two between 32 and 8192. Following the table results in incorrect values (scaled by 1/2) if an RFFT is followed by RIFFT. If that is not the intent of the table, then it may be useful to clarify the inclusion of the additional shift. No shifting is required for the floating point Real FFT, and inversion results in signal recovery as-is.

christophe0606, 2022-02-02:
@tluliak Thank you.
You're right. It is an error in the documentation.

escherstair, 2022-04-11:
Hi,
this topic has been already discussed in #220 and #810 
Probably not 100% fixed

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606