ARM-software/CMSIS_5 #1394: CMSIS-NN basic_test for test_arm_convsolve_s8.c

jiaodawuyanzu, 2022-01-11:
In the CMSIS-NN test case, why is the conv_params.input_offset = BASIC_INPUT_OFFSET parameter 128 in the basic_arm_convolve_s8 function in the test_arm_convsolve_s8.c file? From the input.txt file in the basic subfolder of the PregeneratedData folder, I correspond to the input_data.h file of the test case. Shouldn't onv_params.input_offset be 0? Why is 128?How does the value of Q correspond to the value of R?

felix-johnny, 2022-01-11:
@jiaodawuyanzu Thanks for the question. CMSIS-NN follows symmetric quantization(values centered around 0). But in reality, your input and output data might not be so perfectly centered around 0 and come with an offset(around 0). The [input and output offsets](https://github.com/ARM-software/CMSIS_5/blob/7bb14d86eaa4eff757672705296bf0fe563007e3/CMSIS/NN/Include/arm_nnfunctions.h#L288) define that.  CMSIS-NN follows the TensorFlow Lite's quantization scheme which defines how the offsets are calculated.

 I am not sure sure what you mean by Q and R. could you clarify that?

jiaodawuyanzu, 2022-01-17:
@felix-johnny Thank you for your answer. Q represents the quantized value and R represents the floating point number of the true value. My question is in the test case basic_arm_convolve_s8() function. The input basic_input array is const q7_t basic_input[40] = {-9, -38, 76, -40, 99, 60, 64, 52, 47, -43, 72, 28, -82, 61,-63, 48, -21, -47, 74, 29, -28, 8, -38, 13, 33, -45, 59, -25,-8, -50, 107, 81, -77, 43, -82, 125, 83, 23, 17, 15};, but the actual R data is in the input.txt file in the PregeneratedData directory. The value is -9.000000000000000000e+00.
-3.800000000000000000e+01
7.600000000000000000e+01.... In this way, S = 10 and Z = 0 are calculated using the quantization scheme of TensorFlow Lite. However, why is conv_params.input_offset = BASIC_INPUT_OFFSET in the test function of basic_arm_convolve_s8()?It's got a value of 128, and I understand that it's not a zero?

felix-johnny, 2022-01-17:
@jiaodawuyanzu The input offset is specified by TFLM's quantization [specification](https://www.tensorflow.org/lite/performance/quantization_spec). For the test cases it is generated in this python file https://github.com/ARM-software/CMSIS_5/blob/1cdd0e824db7340b543a7806e9a73244e73c675e/CMSIS/NN/Tests/UnitTest/generate_test_data.py#L888