ARM-software/CMSIS_5 #1396: about develop cortex-a53 kernel

45230, 2022-01-15:
How to develop bare-metal code with cortex-A53 core chips? I have an `I.MX8 mini`.

When I downloaded CMSIS5, I couldn't find the startup code for Cortex-A53 in a folder like ARMCA5.

Can ARMv8MML (or ARMv8MBL, ARMv81MML) be used with the Cortex-A53 kernel?

I am new to cortex-A53 development, please help me, thanks

flit, 2022-01-16:
CMSIS supports M-profile and certain v7-A cores. As the names imply, ARMv8M* variants are for v8-M.

General architecture support questions should be directed to the [Arm Community](https://community.arm.com).
Also, since you are using an NXP product, please seek help from NXP directly. NXP's [community](https://community.nxp.com) is probably a good resource to start with.

JonatanAntoni, 2022-01-17:
There is a pending PR addressing this, see #946.
So far, we do not have official plans to add AARCH64 support to CMSIS.

45230, 2022-01-17:
> CMSIS supports M-profile and certain v7-A cores. As the names imply, ARMv8M* variants are for v8-M.
>
> General architecture support questions should be directed to the [Arm [...]

> There is a pending PR addressing this, see #946. So far, we do not have official plans to add AARCH64 support to CMSIS.

Thank you for your answer

closed 2022-01-26 by 45230