ARM-software/CMSIS_5 #1398: CMSIS-NN asserts on fully connected layer with more than 2 outputs (new in TF 2.7?)

janjongboom, 2022-01-17:
Since switching from TensorFlow 2.4 to 2.7 for training our models we get the following op inserted in a MobileNetV2 model with 0.35 alpha:

[image]

This is incompatible with CMSIS-NN as it can only handle fully connected layers with two outputs, as per https://github.com/tensorflow/tflite-micro/blob/9414ed3266c1ce55033cf3ed7a307881d2531400/tensorflow/lite/micro/kernels/cmsis_nn/fully_connected.cc#L82

As these are our own models we can force the FullyConnected layer to output a 9x16 layer instead (as it did on TensorFlow 2.4) but we can't control the models that our (and Arm's) customers generate, so I think this is a bug worth fixing.

Here you'll find the same model trained with TF2.4 and TF2.7 after quantization: [jan-vs-niet-jan.zip](https://github.com/ARM-software/CMSIS_5/files/7880758/jan-vs-niet-jan.zip) - the only difference I can see is the FullyConnected layer on location 66 (and a 2.0.0 runtime for the file over 1.5.0).

felix-johnny, 2022-01-17:
@janjongboom Thanks for bringing this up. The existing behavior is inline with what is expected of TFLM's reference kernels(https://github.com/tensorflow/tflite-micro/blob/963a47a64dc2c0e021b882c328d43ca45fe3f9d7/tensorflow/lite/kernels/internal/reference/integer_ops/fully_connected.h#L36).  Do you have a ticket on TFLM on this or know the plan there is? 
 

Here are a couple of solutions on how we can handle this in CMSIS-NN.

**Solution 1:**
 The least effort solution is to fold the W & H into N and pass the input tensor as N=9, W=H=1 and C=1280 and ask for an output of 9x1x1x16(this would be in micro/kernels/cmsis-nn/fully_connected.cc). It doesn't matter that CMSIS-NN thinks the output is 9x1x1x16. It is the same as 1x3x3x16 for NHWC format. The subsequent layers can address this as 1x3x3x16 and it should be ok.

  The downside of that change is that we miss out on  performance gains that can be achieved from memory access optimization(it is a read of input from sram vs weights from flash tradeoff ). 

For this particular case and chosen solution (9x1x1x16 and no change to existing FC implementation in CMSIS-NN) you'll get a better performance by replacing the FC layer by a 1x1 conv. But, I guess that is not in your hands! :-) 
 
**Solution 2:**
1. Invoke the 1x1 convolution function instead of  FC in the TFLM CMSIS-NN glue for cases were H or W is not equal to 1. This would give a better performance than solution 1. I have to say here, I am not sure if there is any piece of code in the TFLM Prepare() process that expects W&H as 1.  That is to be checked.

 I would prefer solution 2 as it has better performance without any implicit conversions. Though, I would have really liked it if the model generation process chose a conv layer there instead of FC :-)

felix-johnny, 2022-01-19:
Thinking about Solution 2, got a few other aspects to consider..FC has a per tensor quantization compared to conv which has a per channel quantization. Not sure if the same behavior can be achieved from the model generation process(i.e same quantization params for all channels). 

And if we handle this in the glue i.e, repurpose the FC as a 1x1 conv, then there might be some cases where it would not scale well.. In this case, the scratch memory increase to 16 channels(8 bytes/channel for quantization) will not make a big dent , but if the output channel is large, it would result in a larger scratch memory. 
 Solution 1 might be a better fit considering this.

dansitu, 2022-01-20:
For anyone looking into 1x1 conv as a workaround, an option was recently added to the TF Lite Converter to disable per-channel quantization for a model:

https://github.com/tensorflow/tensorflow/commit/1b1833e939b6fdb4c11973fdc506e4081a2d7c25#diff-dd8b67d882b4f2c3d915602a68548db658522449998dd1520c546fc2da31ba0d

Note that currently this is global; it would be great if it could be specified per-layer.