ARM-software/CMSIS_5 #1400: Building CMSIS DSP extensions for Cortex-M3

elagil, 2022-01-18:
Hello,

in trying to build CMSIS-DSP for Cortex-M3 with CMake, I stumble across an issue, where an environment variable `ARMCM3` does not seem to be defined. Building works for many other cores, e.g. M0, M33, M4, M7, ...

Initially, I specify my target platform and core with

```
set(PLATFORM "FVP")
set(ARM_CPU "cortex-m3")
```

Now, following the hierarchy of CMake, I find that in the file https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/configPlatform.cmake, there is no option for the Cortex-M3. If I add it manually to the file like this:

```
  ###################
  #
  # Cortex cortex-m3
  #
  if (ARM_CPU MATCHES "^[cC]ortex-[mM]3([^0-9].*)?$")
      SET(CORE ARMCM3 PARENT_SCOPE)
  endif()
```

it builds fine. Is this omission a bug or intentional?

Thanks in advance!

christophe0606, 2022-01-18:
@elagil This set of cmake files is mainly used for our CMSIS-DSP test framework. So it is supporting a limited set of cores (enough to have the needed coverage).

The official way to build is through CMSIS Packs. And you can also generate a Cmake (a simpler one) from CMSIS Packs using cmsis-build.

That being said : the fix you did look right. 
It is not an intentional omission but the M3 target was never used through this set of cmake that's why it is missing.

elagil, 2022-01-18:
@christophe0606 Thanks, I see! I understand that I need to install the `cbuild` toolkit, using `cbuild_install`. Then, I can use some *.cprj file with `cbuildgen`. 

Is there an example for such a file, for only compiling the DSP portion of CMSIS? Or better, is there a tool for creating this file? The options for settings in this file format seem overwhelming.

elagil, 2022-01-18:
@JonatanAntoni Thanks, I will have a look!

closed 2022-01-18 by elagil