ARM-software/CMSIS_5 #1401: Building with CMake and GCC

elagil, 2022-01-18:
Hello!

I want to utilize CMSIS (currently, especially the DSP component), while building with CMake and GCC. In the near future, I want to use other CMSIS components as well. The issue is the creation of a suitable `CMakeLists.txt`, which defines the build options and included libraries.

As suggested in https://github.com/ARM-software/CMSIS_5/issues/1400, I shall define a project by means of a `*.cprj` file, whose format is described in https://arm-software.github.io/CMSIS_5/Build/html/cprjFormat_pg.html. Then, I can use this file with the `cbuildgen` tool, which will output  the desired  `CMakeLists.txt`. Finally, I can build the project.

How can I generate this `*.cprj`? I am under the impression that I may use the Keil MDK, create a project there (some `*.uvprojx` format?) and convert it by means of `cbuild_uv.sh`. Is there a simpler way? The problem is, I do not have access to Keil MDK and I am not sure, if I am capable of writing the `*.cprj` file from scratch.

@brondani, it was suggested that you may have some information, being the build expert.

Thanks in advance!

brondani, 2022-01-27:
Hi @elagil,

You can use Keil MDK to create a project and [export](https://www.keil.com/support/man/docs/uv4/uv4_ui_export.htm) it in the CPRJ format, even though the export mechanism is not complete for GCC projects and you may need to manually adapt/correct it anyway.

In the Open-CMSIS-Pack/devtools repository you can find [GCC examples](https://github.com/Open-CMSIS-Pack/devtools/tree/main/tools/buildmgr/test/testinput/Examples/GCC) that you can use as inspiration for handcrafting your own CPRJ files.

Alternatively as part of the Open-CMSIS-Pack initiative we are working on a tool for describing projects in an easier YAML based format and generating CPRJ files from them. It is still experimental, if you want to try it we should publish an alpha release in a few days or you can build it from sources. In the [projmgr overview](https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Overview.md) you can find the documentation draft.

Cheers
Daniel

elagil, 2022-02-06:
@brondani Thanks a lot for your insight! I will most likely pursue the last approach, which looks very promising.

Best regards
Adrian

closed 2022-02-06 by elagil

tobermory, 2022-02-22:
I build CMSIS components (RTOS2, RTX, not currently DSP) using just make/GCC, so not CMake but at least with GCC. You might want to look at

https://github.com/tobermory/RTX-make-gcc

elagil, 2022-02-28:
@tobermory Thanks, I will.