ARM-software/CMSIS_5 #1402: Cast-align warning and error with -Wcast-align

myronvsqm, 2022-01-19:
https://github.com/ARM-software/CMSIS_5/blob/4946da8d01f3245182c3e6094b9da1fcd1070269/CMSIS/Core/Include/cmsis_gcc.h#L200

JonatanAntoni, 2022-01-20:
Hi @myronvsqm,

My I ask you to check if the following alternative implementation would work for you, please?

```C
__STATIC_FORCEINLINE void __TZ_set_STACKSEAL_S (uint32_t* stackTop) {
  stackTop[0] = (uint32_t)(__TZ_STACK_SEAL_VALUE);
  stackTop[1] = (uint32_t)(__TZ_STACK_SEAL_VALUE >> 32);
}
```

Thanks,
Jonatan

GuentherMartin, 2022-01-20:
Hi @myronvsqm,

May I ask you which GCC version and which optimization level you are using and what kind of error you get?

Thank you,
Martin

myronvsqm, 2022-01-20:
> Hi @myronvsqm,
>
> My I ask you to check if the following alternative implementation would work for you, please?
>
> ```c
> __STATIC_FORCEINLINE void __TZ_set_STACKSEAL_S (uint32_t* stackTop) { [...]

Hi @JonatanAntoni ,

This works thank you. Please fix in your master.

Thanks

Regards,
Myron

myronvsqm, 2022-01-20:
> Hi @myronvsqm,
> 
> May I ask you which GCC version and which optimization level you are using and what kind of error you get?
> 
> Thank you, Martin

Hi Guenther,

Any GCC compiler with -Wcast-align flag included in the build I believe.
-Werror = cast-align.