ARM-software/CMSIS_5 #1411: arm_fully_connected_mat_q7_vec_q15_opt failing to build in IAR due to inline asm

yair-ehrenwald, 2022-02-03:
Fails with this error:
Error[Be030]: unsupported inline assembler argument reference: operand index missing

                        "bne COL_LOOP_%=\n"

Any pointers on how to get it running in IAR? Using a GCC based toolchain I got better results than with the intrinsics.

yair-ehrenwald, 2022-02-03:
Thanks for the quick reply :)

The USE_INTRINSIC define work. However, from some testing on a similar platform (both Cortex-M4F) but with a GCC based 
toolchain, the inline assembly version gives a much better cycle count which is critical is this case.

I assume the issue is a matter of different syntax between GCC and IAR assembly? The only info I found on this is:
https://www.iar.com/knowledge/support/technical-notes/compiler/migrating-some-gcc-inline-assembler-constructions-to-ewarm/
Which wasn't helpful enough :) If you have any pointers I'd be glad to do this myself.

I'm familiar with TFLM, unfortunately it's not suitable for this project.

Thanks,
Yair

yair-ehrenwald, 2022-02-05:
Played around with the code a little. Turns out the syntax differences are very small :)
 Just in case anybody else runs into this, remove the "%" from the loop label and for the subs instruction don't use shorthand: subs %[colCnt],%[colCnt], #1 instead of subs %[colCnt], #1 

Yair

felix-johnny, 2022-02-06:
@yair-ehrenwald Thanks for the update!

closed 2022-02-06 by felix-johnny