ARM-software/CMSIS_5 #1412: Python Wrapper memory leak

antoine-bernabeu, 2022-02-04:
It appears that when using the Python wrapper, there is some kind of memory leak.
For example I can fill up all my RAM just by doing the same FFT.
Code to reproduce: 
```
data = np.random.normal(loc = 0, scale = 1, size = 256)
data_q15 = arm_float_to_q15(data)
rfft_instance_q15 = arm_rfft_instance_q15()
status = arm_rfft_init_q15(rfft_instance_q15, 256, 0, 1)
while True:
    fft_data = arm_rfft_q15(rfft_instance_q15, data_q15)
```

christophe0606, 2022-02-04:
@antoine-bernabeu Thank you.
I'll have a look (but not in short term since my bandwidth is currently taken on other topics).
It is something I wanted to analyze more : the memory management of the wrapper. 
So the reproducer will be useful.

christophe0606, 2022-02-09:
@antoine-bernabeu The latest commit e04e03a5d7b4a5a7c2ef35fbe0e7d0cd18915f2b  should solve the issue I hope.

Can you confirm ?

antoine-bernabeu, 2022-02-09:
Yep, I confirm.
Thank you for fixing this :+1:

closed 2022-02-09 by christophe0606