ARM-software/CMSIS_5 #1413: wrong initial stack pointer in the startup_ARMCM33.s

JacobPeng, 2022-02-07:
[image]

According to the Arm®v8-M Architecture Reference Manual, the initial stack pointer of Cortex M33 should be msp (**main stack pointer**) instead of psp.

So the code in the startup_ARMCM33.s should be
```
Reset_Handler:
                ldr      r0, =__INITIAL_SP
                msr      msp, r0  /* or mov sp, r0 */
```
instead of
```
Reset_Handler:
                ldr      r0, =__INITIAL_SP
                msr      psp, r0
```

JonatanAntoni, 2022-02-07:
Hi @JacobPeng,

Please be aware that the Arm M architecture guarantees that the `MSP` is automatically initialized by hardware.
The additional initialization of `PSP` during Startup happens to assure that it doesn't stay uninitialized. The stack sealing applied to `MSP` is reused by setting `PSP` to the same address.

Of course, one must not use the process stack without properly initializing a dedicated memory area to it. This is typically done by an RTOS after reaching user main.

Does this clarify your question?

Cheers,
Jonatan

JacobPeng, 2022-03-01:
> Please be aware that the Arm M architecture guarantees that the MSP is automatically initialized by hardware.  

```
Reset_Handler:
                ldr      r0, =__INITIAL_SP
                msr      psp, r0
```
The upper code in the startup_ARMCM33.s is reasonable if MSP is initialized by hardware. 

Q1: Do you mind sharing the relevant publishing documentations? I didn't see anything about it in the Arm®v8-M Architecture Reference Manual.

Q2: How does the hardware initialize MSP to the right value? Because stack pointers are different in the different projects, it is difficult for hardware to get the right stack pointer.

Q3: Can the vendor modify the default behavior of MSP that is not initialized by hardware? I found that one CM33 chip did not initialize MSP after power reset.

JonatanAntoni, 2022-03-01:
> Q1: Do you mind sharing the relevant publishing documentations? I didn't see anything about it in the Arm®v8-M Architecture Reference Manual.

You can find the latest Armv8-m ARM at https://developer.arm.com/documentation/ddi0553/latest.
In issue R the `TakeReset` pseudocode is in section E2.1.395 on page 2081.
 
> Q2: How does the hardware initialize MSP to the right value? Because stack pointers are different in the different projects, it is difficult for hardware to get the right stack pointer.

The stack pointer is taken from the vector table, see pseudo code line 49. Depending on running in secure or non-secure mode this value is used to initialize MSP_S (line 57) or MSP_NS (line 61).
 
> Q3: Can the vendor modify the default behavior of MSP that is not initialized by hardware? I found that one CM33 chip did not initialize MSP after power reset.

No and yes. The default behaviour cannot be changed but could be overwritten by some hidden firmware (such as a builtin bootloader). In this case you might recognize different core register values when execution reaches the reset handler. I'd consider this a "bug" in the chip and you'd need to work around it by modifying the startup code approtiately.

JacobPeng, 2022-03-02:
Thank you very much for the very detailed response. I understand it now. I will try to contact the vendor.

closed 2022-03-02 by JacobPeng